cta-observatory/ctapipe #297: scikit-hep, histogrammar relevant for us?

cdeil, 2017-01-11:
I just now learned of https://github.com/histogrammar/histogrammar-python . Apparently there's a lot of Python activity in HEP and e.g. https://github.com/scikit-hep/scikit-hep is starting.

In Gammapy, @joleroi re-wrote the NDDataArray class last week (see [nddata_demo.ipynb](https://github.com/gammapy/gammapy-extra/blob/master/notebooks/nddata_demo.ipynb)). At the moment our `NDDataArray` only represents n-dim data array with axes that have some binning, and interpolation. But no fill yet or other histogram operations as available in http://histogrammar.org/docs/specification/1.0/ .

@ParsonsRD @kosack @joleroi @TarekHC  - `histogrammar-python` might be useful for ctapipe / Gammapy to use directly or study as an example how to implement histograms? I think it would be nice if we can find a way to collaborate or share development work on this basic data structure.

kosack, 2017-03-13:
there's also `matplotlib-hep`  (https://github.com/ibab/matplotlib-hep) which also provides better histogram plotting.  I would like to avoid adding to many dependencies, however - we already exceed the number we wanted, so it has to be evaluated whether or not this is needed over a simple wrapper class for np.histogramdd to meet our needs.  We have very little use for `nddata` for example, except perhaps for IRF generation if that is done in this framework and not the science tools.

kosack, 2017-03-13:
Personally, I dislike `histogrammer`'s interface, since it is not very pythonic (and much more ROOT-like). It's also quite slow, since it requires an interpreter loop to fill.  I also think we should not be making a lot of histograms inside the ctapipe code. Rather, we should output tabular files (event-lists, etc), and make histograms on them separately. That is different than the ROOT-way-of-life, but is more flexible and faster, and avoids the histogram-bloat that occurs in current implementations.  The HESS (and HEP) philosophy of  "everything is a histogram" is really not a good design.

cdeil, 2017-03-13:
@kosack - I also looked a bit and agree that mostly likely we don't want to user their code.

But there's still the question whether we should build a nice histogram class, and what features is should have (e.g. under / overflow bins, serialisation, bin values and errors?, rebinning?).
So ROOT and histogrammer it might be a useful example to look at once we implement our own code that builds IRFs from events.

I'm not sure keeping this issue open as a reminder that histogrammar exists is useful or not.
@kosack - Maybe close now?

kosack, 2017-03-15:
Yes, I think a nicer histogram class (or histogram manager class for collections of histograms) is a separate issue that was already brought up in internal meetings,  Right now `ctapipe.utils.Histogram` is sufficient for simple things, but we could certainly improve it with errors, etc.

closed 2017-03-15 by kosack