cta-observatory/ctapipe #319: ShowMaxEstimator is missing an input file

kosack, 2017-01-25:
@tino-michael  It seems there is a file missing in ctapipe-extra repo.  Can you add it and check it into ctapipe-extra?

```
self = <ctapipe.reco.shower_max.ShowerMaxEstimator object at 0x23ae10dd8>
filename = '/Users/kosack/Projects/CTA/Working/ctapipe/ctapipe-extra/datasets/atmprof_paranal.dat', col_altitude = 0, col_thickness = 2

    def __init__(self, filename, col_altitude=0, col_thickness=2):
        """
            small class that calculates the height of the shower maximum
            given a parametrisation of the atmosphere
            and certain parameters of the shower itself
    
            Parameters
            ----------
            filename : string
                path to text file that contains a table of the
                atmosphere parameters
            col_altitude : int
                column in the text file that contains the altitude/height
            col_thickness : int
                column in the text file that contains the thickness
            """
    
        altitude = []
        thickness = []
>       atm_file = open(filename, "r")
E       FileNotFoundError: [Errno 2] No such file or directory: '/Users/kosack/Projects/CTA/Working/ctapipe/ctapipe-extra/datasets/atmprof_paranal.dat'

ctapipe/reco/shower_max.py:30: FileNotFoundError
```

kosack, 2017-01-25:
actually, it seems that file is in `ctapipe-extras/datasets` which was renamed `ctapipe-extras/ctapipe_resources`, so it just needs to be moved

kosack, 2017-01-25:
so never mind, I'll fix it and remove ctapipe-extra as a sub-module (it will be installed as a package)

tino-michael, 2017-01-25:
The test script itself uses `get_path("atmprof_paranal.dat")`, so it's completely agnostic about where the file actually resides.

kosack, 2017-01-25:
yes, it's related to the fact that the file is not in the sub-module version of ctapipe-extra for some reason (or at least get_path fails to find it).

kosack, 2017-01-25:
Submodules seem to get out of sync easily and are difficult to maintain anyhow, so I'm switching it to a package.

tino-michael, 2017-04-04:
can this be closed?

kosack, 2017-04-10:
it's still not in the linked ctapipe_extras dir, but this will be closed once I finish making it a separate package, rather than a sub-repo as in #222  (which is partially complete, but not yet tested)

closed 2017-05-04 by kosack