cta-observatory/ctapipe #332: Astropy v1.3 breaks coordinate transformations

ParsonsRD, 2017-02-17:
Hi all,

I just checked out the latest version of astropy and it seems version 1.3 has made some changes to the coordinate implementation which breaks the ctapipe code (error below). I'll look into this next week and try to come up with a solution.

```
> ---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
/home/puerck/anaconda3/envs/cta/lib/python3.5/site-packages/astropy/coordinates/baseframe.py in __get__(self, instance, frame_cls)
   228                     else:
--> 229                         out = np_broadcast_to(out, instance_shape, subok=True)
   230                 except ValueError:

/home/puerck/anaconda3/envs/cta/lib/python3.5/site-packages/numpy/lib/stride_tricks.py in broadcast_to(array, shape, subok)
   114     """
--> 115     return _broadcast_to(array, shape, subok=subok, readonly=True)
   116

/home/puerck/anaconda3/envs/cta/lib/python3.5/site-packages/numpy/lib/stride_tricks.py in _broadcast_to(array, shape, subok, readonly)
    69         (array,), flags=['multi_index', 'refs_ok', 'zerosize_ok'] + extras,
---> 70         op_flags=[op_flag], itershape=shape, order='C').itviews[0]
    71     result = _maybe_view_as_subclass(array, broadcast)

ValueError: operands could not be broadcast together with remapped shapes [original->remapped]: (2,) and requested shape (1855,)

During handling of the above exception, another exception occurred:

ValueError                                Traceback (most recent call last)
/home/puerck/ctasoft/ctapipe/textscript.py in <module>()
   114
   115 if __name__ == "__main__":
--> 116     test_ImPACT_fit()

/home/puerck/ctasoft/ctapipe/textscript.py in test_ImPACT_fit()
    76                                        rotation= -1* cam_geom[tel_id].cam_rotation )
    77             print(nom_system)
---> 78             nom_coord = camera_coord.transform_to(nom_system)
    79
    80             mask = tailcuts_clean(cam_geom[tel_id], pmt_signal, 1,

/home/puerck/anaconda3/envs/cta/lib/python3.5/site-packages/astropy/coordinates/baseframe.py in transform_to(self, new_frame)
   986             msg = 'Cannot transform from {0} to {1}'
   987             raise ConvertError(msg.format(self.__class__, new_frame.__class__))
--> 988         return trans(self, new_frame)
   989
   990     def is_transformable_to(self, new_frame):

/home/puerck/anaconda3/envs/cta/lib/python3.5/site-packages/astropy/coordinates/transformations.py in __call__(self, fromcoord, toframe)
   905
   906             curr_toframe = t.tosys(**frattrs)
--> 907             curr_coord = t(curr_coord, curr_toframe)
   908
   909         # this is safe even in the case where self.transforms is empty, because

/home/puerck/anaconda3/envs/cta/lib/python3.5/site-packages/astropy/coordinates/transformations.py in __call__(self, fromcoord, toframe)
   705
   706     def __call__(self, fromcoord, toframe):
--> 707         res = self.func(fromcoord, toframe)
   708         if not isinstance(res, self.tosys):
   709             raise TypeError('the transformation function yielded {0} but '

/home/puerck/ctasoft/ctapipe/ctapipe/coordinates/angular_frames.py in camera_to_telescope(camera_coord, telescope_frame)
   408     representation = PlanarRepresentation(x_rotated, y_rotated)
   409
--> 410     return telescope_frame.realize_frame(representation)
   411
   412

/home/puerck/anaconda3/envs/cta/lib/python3.5/site-packages/astropy/coordinates/baseframe.py in realize_frame(self, representation)
   876                         for attr in self.get_frame_attr_names()
   877                         if attr not in self._attr_names_with_defaults])
--> 878         return self.__class__(representation, **frattrs)
   879
   880     def represent_as(self, new_representation, in_frame_units=False):

/home/puerck/anaconda3/envs/cta/lib/python3.5/site-packages/astropy/coordinates/baseframe.py in __init__(self, *args, **kwargs)
   660                 # Validate attribute by getting it.  If the instance has data,
   661                 # this also checks its shape is OK.  If not, we do it below.
--> 662                 values[fnm] = getattr(self, fnm)
   663             else:
   664                 setattr(self, '_' + fnm, fdefault)

/home/puerck/anaconda3/envs/cta/lib/python3.5/site-packages/astropy/coordinates/baseframe.py in __get__(self, instance, frame_cls)
   233                         "attribute {0} should be scalar or have shape {1}, "
   234                         "but is has shape {2} and could not be broadcast."
--> 235                         .format(self.name, instance_shape, out.shape))
   236
   237                 converted = True

ValueError: attribute pointing_direction should be scalar or have shape (1855,), but is has shape (2,) and could not be broadcast.
```

kosack, 2017-03-13:
For now, I've updated the conda package to require a lower version of `astropy`, which also means `ctapipe` is not yet compatible with python 3.6, which doesn't have an astropy 1.2 release until this is fixed.

ParsonsRD, 2017-04-18:
Should be fixed as of merge #364

closed 2017-04-18 by ParsonsRD