cta-observatory/ctapipe #346: proposal to speed up algorithms that use pixel neighbor lists

kosack, 2017-03-15:
I did a study of why the image cleaning algorithm (tailcuts) is slower than I would like, and found the issue is how the neighborlist is stored and applied.  The following shows several implementations of the tailcuts algorithm, using different representations of the neighbor list.  (conclusion below)

[image]

The conclusion is that using a 2D boolean matrix for the neighbors and *numpy* operations is significantly faster than even a simple for-loop based C implementation (the *cython-opt* method is equivalant to a c for-loop method with compiler optimizations turned on). The reason is likely the vectorizability of operations and modern CPU features.  Using the following neighbor matrix instead of an adjacency list of pixel-ids seems to be the fastest improvement:

[image]

Using this will likely also allow for speedups of other algorithms.  I plan on implementing it directly in the code in a future pull-request.

Full study here:
https://github.com/kosack/ctapipe/blob/improve_cleaning_speed/examples/notebooks/speed_up_image_cleaning.ipynb

kosack, 2017-03-15:
@watsonjj this could likely also speed up parts of the calibration that use nearest-neighbors. Though, it means some re-structuring of code that uses them.

watsonjj, 2017-04-04:
I've seen this, and will test/implement it in the charge extraction techniques as soon as I have time.

kosack, 2017-04-04:
Great, I'll modify the `CameraGeometry` class and the `image` module cleaning functions to use it, then you don't have to do much other than update your algorithms.

kosack, 2017-04-06:
@watsonjj once  PR #360 is merged, you can use the `geom.neighbor_matrix` attribute to speed up your algorithms (see `image.tailcuts_clean` and `image.dilate` for examples).  It may also simplify some other code, since you no longer need to cache the camera geometries youself, you can just keep loading or calling `.guess`, which is much faster now and automatically caches previously constructed geometries.

watsonjj, 2017-04-06:
@kosack I will try to look at this today. A problem I previously had with caching was if I was working with multiple files, as I was caching by telescope index. This is much nicer!

closed 2017-04-07 by kosack