cta-observatory/ctapipe #356: Tracebacks are hidden when occurring inside a Tool

watsonjj, 2017-04-04:
When an error is raised during the execution of a Tool, the traceback is not shown. This also seems to be the case with --log-level=DEBUG.

kosack, 2017-04-06:
That is just because exception is caught somewhere. I think that's actually reasonable behavior - any exception should be handled somewhere in the tool. Any ones that are not handled will then end up in the default python exception handler that shows the traceback.  To see the traceback, you can always use a debugger, or we could make it an option (e.g. if debug is the log-level or something) to dump it.

kosack, 2017-04-06:
My guess is it's being caught somewhere in the traitlets.config.Application base class, since the Tool class doesn't do it.

closed 2017-06-07 by kosack