cta-observatory/ctapipe #363: Improving the NeighbourPeakIntegrator

watsonjj, 2017-04-12:
The new neighbor_matrix was recently added to the CameraGeometry class, which could potentially improve the speed of algorithms that require information about the neighbours to the pixel.

I have attempted to improve NeighbourPeakIntegrator by using the new neighbor_matrix, but the most obvious implementation using it is 10x slower than the original method using a list.

Below is a link to an IPython notebook containing a comparison between the old method and the new method. Can anyone find a better implementation of the algorithm that will improve the speed?

https://www.dropbox.com/s/omy60lnw3su6iy5/new_NeighbourPeakIntegrator.ipynb?dl=0

ParsonsRD, 2017-04-13:
Hi, I think the problem here is that using the neighbour matrix you are allocating an array of size 1855 * 1855 * 30 * 1. i.e. more than 100 million entries.

I think no matter what you do with such long arrays the algorithm will be slow, simply due to the memory allocation and the filling of the array!

kosack, 2017-04-18:
You shouldn't need to generate your own neighbor matrix, just use the one in the CameraGeometry (there is no reason to re-generate it).  it also should not have to be 4 dimensional - broadcasting rules should work fine.

kosack, 2017-04-18:
It's actually maybe a good case for cython, since it's a complex operation that can easily be done with a loop.

ParsonsRD, 2017-04-18:
Could we try using this as an example next week?

kosack, 2017-04-18:
sure, it would be a nice example of cython maybe.

btw, to use cython in a notebook is really easy:

```
%load_ext cython
```

then any cell that starts with `%%cython` gets compiled automatically:

```python
%%cython

def cython_func(x):
    cdef int ii
    cdef double y = 0
    for ii in range(100):
         y += ii
    return y

ParsonsRD, 2017-04-18:
Oh wow, I wish I knew that earlier :-)

closed 2017-06-19 by kosack