cta-observatory/ctapipe #370: List HDF5 and pytables in environment

cdeil, 2017-04-22:
I was trying to get set up for ctapipe development and ran into this error:
```
$ make develop
CTAPIPE_EXTRA_DIR=/Users/deil/code/ctapipe/ctapipe-extra python setup.py develop
running develop
running egg_info
writing ctapipe.egg-info/PKG-INFO
writing requirements to ctapipe.egg-info/requires.txt
writing entry points to ctapipe.egg-info/entry_points.txt
writing top-level names to ctapipe.egg-info/top_level.txt
writing dependency_links to ctapipe.egg-info/dependency_links.txt
reading manifest file 'ctapipe.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no files found matching 'CHANGES.rst'
warning: no files found matching 'ctapipe/VERSION'
warning: no files found matching '*.c' under directory '*.pyx'
warning: no files found matching '*.pxd' under directory '*.pyx'
warning: no files found matching '*' under directory 'scripts'
no previously-included directories found matching 'build'
no previously-included directories found matching 'docs/_build'
no previously-included directories found matching 'docs/api'
warning: no previously-included files matching '*.pyc' found anywhere in distribution
warning: no previously-included files matching '*.o' found anywhere in distribution
writing manifest file 'ctapipe.egg-info/SOURCES.txt'
running build_ext
Creating /Users/deil/software/anaconda/envs/cta-dev/lib/python3.5/site-packages/ctapipe.egg-link (link to .)
Adding ctapipe 0.4.0.post105 to easy-install.pth file
Installing ctapipe-chargeres-extract script to /Users/deil/software/anaconda/envs/cta-dev/bin
Installing ctapipe-info script to /Users/deil/software/anaconda/envs/cta-dev/bin
Installing ctapipe-camdemo script to /Users/deil/software/anaconda/envs/cta-dev/bin
Installing ctapipe-flow script to /Users/deil/software/anaconda/envs/cta-dev/bin
Installing ctapipe-chargeres-plot script to /Users/deil/software/anaconda/envs/cta-dev/bin
Installing ctapipe-dump-triggers script to /Users/deil/software/anaconda/envs/cta-dev/bin
Installing ctapipe-chargeres-hist script to /Users/deil/software/anaconda/envs/cta-dev/bin

Installed /Users/deil/code/ctapipe
Processing dependencies for ctapipe==0.4.0.post105
Searching for tables
Reading https://pypi.python.org/simple/tables/
Downloading https://pypi.python.org/packages/a2/e5/697f7ec96c4808983711635ef2e4c6b217493343938410125bcece1642cf/tables-3.4.2.tar.gz#md5=dfb757740a915f34f8f2cf618b7eb2bf
Best match: tables 3.4.2
Processing tables-3.4.2.tar.gz
Writing /var/folders/sb/4qv5j4m90pz1rw7m70rj1b1r0000gn/T/easy_install-cg5507y0/tables-3.4.2/setup.cfg
Running tables-3.4.2/setup.py -q bdist_egg --dist-dir /var/folders/sb/4qv5j4m90pz1rw7m70rj1b1r0000gn/T/easy_install-cg5507y0/tables-3.4.2/egg-dist-tmp-_gamj3o9
* Using Python 3.5.1 |Continuum Analytics, Inc.| (default, Jun 15 2016, 16:14:02) 
* USE_PKGCONFIG: True
* Found conda env: ``/Users/deil/software/anaconda/envs/cta-dev``
/var/folders/sb/4qv5j4m90pz1rw7m70rj1b1r0000gn/T/easy_install-cg5507y0/tables-3.4.2/temp/H5closednkrs00y.c:1:1: warning: type specifier missing, defaults to 'int'
      [-Wimplicit-int]
main (int argc, char **argv) {
^
/var/folders/sb/4qv5j4m90pz1rw7m70rj1b1r0000gn/T/easy_install-cg5507y0/tables-3.4.2/temp/H5closednkrs00y.c:2:5: warning: implicit declaration of function 'H5close' is
      invalid in C99 [-Wimplicit-function-declaration]
    H5close();
    ^
2 warnings generated.
ld: library not found for -lhdf5
clang: error: linker command failed with exit code 1 (use -v to see invocation)
.. ERROR:: Could not find a local HDF5 installation.
   You may need to explicitly state where your local HDF5 headers and
   library can be found by setting the ``HDF5_DIR`` environment
   variable or by using the ``--hdf5`` command-line option.
error: Setup script exited with 1
make: *** [develop] Error 1
```
It looks like the conda end I got doesn't have HDF5?
Is this a peculiarity with my conda setup or something everyone will run into, and the conda environment.yml needs to be amended?

cdeil, 2017-04-22:
I also get this error:
```
(cta-dev) localhost:ctapipe deil$ make test
CTAPIPE_EXTRA_DIR=/Users/deil/code/ctapipe/ctapipe-extra python setup.py test
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: invalid command 'pytest'
make: *** [test] Error 1
```
and this one that psutil is missing:
https://gist.github.com/cdeil/1433d535299564cf1179782d4be7faac

At least for psutil it seems clear that it's simply missing from `environment.yml`, after installing that I get:
```
$ python -m pytest
========================================================================= test session starts ==========================================================================
platform darwin -- Python 3.5.1, pytest-3.0.7, py-1.4.33, pluggy-0.4.0 -- /Users/deil/software/anaconda/envs/cta-dev/bin/python
cachedir: .cache
rootdir: /Users/deil/code/ctapipe, inifile: setup.cfg
collected 145 items / 1 errors 

================================================================================ ERRORS ================================================================================
____________________________________________________________ ERROR collecting ctapipe/io/tests/test_hdf5.py ____________________________________________________________
ImportError while importing test module '/Users/deil/code/ctapipe/ctapipe/io/tests/test_hdf5.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
ctapipe/io/tests/test_hdf5.py:2: in <module>
    from ctapipe.io.hdftableio import SimpleHDF5TableWriter, SimpleHDF5TableReader
ctapipe/io/hdftableio.py:8: in <module>
    import tables
E   ImportError: No module named 'tables'
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! Interrupted: 1 errors during collection !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
======================================================================= 1 error in 5.45 seconds ========================================================================
```

@kosack - Can you fix the conda setup so that it "just works"?

kosack, 2017-05-04:
try it again - I somehow must have reverted the change that added pytables to the default environment (our dependencies are a bit crazy right now, and need to be simplified).

kosack, 2017-05-04:
> I also get this error:
>(cta-dev) localhost:ctapipe deil$ make test

Please check you have the latest version (git pull upstream master), since this was changed at the workshop (it now runs "python -m pytest", which makes it work regardless of what the system command is)

cdeil, 2017-05-07:
I now get this: https://gist.github.com/cdeil/228c77c940996233b53606de01886175

Your CI is broken: https://travis-ci.org/cta-observatory/ctapipe/jobs/228788266#L652
So I don't know if the local test fails I see are a problem with my setup or with ctapipe master.
Don't leave your CI broken all the time, it makes it useless and devs learn not to check or trust it.

Also I had to find out via Google how to update after a while when the `environment.yml` changes. It's simply:
```
conda env update -f environment.yml
```
You could mention that at https://cta-observatory.github.io/ctapipe/getting_started/index.html

kosack, 2017-05-09:
travis and deps listed in environment.yml should all be fixed now. There were a few missing ones.

closed 2017-05-09 by kosack