cta-observatory/ctapipe #404: camera names need unification

kosack, 2017-05-04:
Right now we have the following hard-coded camera names (which are set in `instument/camera.py`) and are also written to `ctapipe_resources`:

- ASTRI (ASTRICam?)
- FlashCam
- NectarCam
- LSTCam
- GCT  (should this be GATE? - it is called both in various places)
- SCTCam (is this the correct name?)
- SST-1m (should be a camera name not a telescope name - is there a correct name? is it DigiCam)
- HESSI - HESS-I cameras

Please suggest updated/better names, since we need to fix them now (the camera name is the "camera_id" used in many parts of the code, and there is now a mismatch e.g. in the muon code)

@AMWMitchell @watsonjj @cocov

watsonjj, 2017-05-04:
GATE is what it was previously called in the pipeline, but I think it should be GCT. GATE is only the telescope structure, and has nothing to do with the camera...

AMWMitchell, 2017-05-04:
Hi,

Agreed - it should be either GCT or CHEC - wondered why it was GATE originally. The muon code might need updating to match. 
SST-1M should probably be DigiCam - as far as I’m aware, ASTRI don’t have any special name for their camera, so either ASTRI or ASTRICam is probably fine. 

Cheers,
Alison

GernotMaier, 2017-05-04:
Is it really necessary to use hardcoded camera names? None of the calibration methods/reconstruction routines depend on the cameras. It is the other way around, and it therefore  seems to be more a configuration issue.

When using name, keep in mind that cameras are still evolving. It is also not guaranteed e.g. that the first GCT camera is the same as the last one (build in 5-6 years? This might matter for the naming, but again it is probably better to have a configuration file ( Tel x -> use methods a,b,c).  

(maybe all of this is already taken into account)

watsonjj, 2017-05-05:
Gernot has a good point. Even now, there is a difference between GCT with the CHEC-M camera, and GCT with the CHEC-S camera. Also combinations between ASTRI and GCT may be used in the near future (CHEC-S on ASTRI structure, ASTRI camera on GCT structure), especially in MC.

watsonjj, 2017-05-05:
Perhaps my comment is an argument to really separate between camera and structure names... in that case there will be a CHECS and CHECM in `instument/camera.py`.

AMWMitchell, 2017-05-05:
However, I don’t think this is a cause for concern regarding Gernot’s first point on the existence of hard-coded camera names; as far as I’m aware the routines are camera independent - the names are used just to call the appropriate configurations (or configuration used to retrieve name) and are only otherwise relevant for e.g. storing cut values, much of which can also eventually be moved to config files. 

The case of changing designs is less clear, as it depends to what extent this is relevant for different parts of ctapipe, (CHEC-M vs S will probably not change the muon cuts, for example) although some way of keeping track and accessing the correct configuration is clearly needed. 

Maybe just “if string contains X, do Y” where multiple designs of a given camera exist. 

RE Jason —> yes, separate camera / structure names are needed, also for the MSTs. 

Cheers,
Alison

watsonjj, 2017-05-05:
The only routines that are camera dependant that come to mind are the R1 calibration. However I think this doesn't change much - currently the r1 calibration is dependant on the r0 reader that is used, and will not be a problem for normal operation where dl0 is read in the common data format.

kosack, 2017-05-18:
There probably aren't many places that need hard-coded names, but I definitely want to be able to use the `CameraGeometry.from_name` functionality for debugging and testing purposes, as well as being able to display the name on any plots, which means there needs to be a unified "camera_id" string.  At least those should match the actual camera names, which they don't right now.  The original idea was that that name would also include a version (or have it as a separate key), so we can support evolving cameras.  E.g. you could have either different names like "CHEC-P" and "CHEC-S", or key them all with an identifier + version `('CHEC', 'S1.0')`.

Right now i have files in the `ctapipe_resources` dir that are keyed by camera_id name only, but that can be expanded to have the version included.

kosack, 2017-05-18:
> Maybe just “if string contains X, do Y” where multiple designs of a given camera exist.

I think that's error prone, so would prefer having a `camera_id` and `camera_version` as separate keys. Then you could say "give me the latest CHEC camera" or "give me the CHEC camera corresponding to a particular date", or "give me CHEC camera version 1.2"

kosack, 2017-05-18:
I think I will change `CameraGeometry.from_name` to ask for a camera_id string + an optional version (which will currently be ignored until we have a real instrument database). I could also imaging loading by date range or obs_id.  This is the same idea for all instrument classes - we need to be able to load them in many ways based on metadata.

AMWMitchell, 2017-05-18:
Sounds good to me - a camera_id + version is probably the best way forward.

kosack, 2017-06-07:
This is basically fixed by #414 (at least it's better than before!)

closed 2017-06-07 by kosack