cta-observatory/ctapipe #411: ImPACT dependencies

tino-michael, 2017-05-18:
It appears the impact module uses iminuit and that it is not part of the defined ctapipe dependencies, so I do not have it installed. Trying to install it with conda throws an error message about conflicting packages:

> ModuleNotFoundError: No module named 'iminuit'
tmichael@sappcr107:~/software/ctapipe/ctapipe/reco/tests$ conda install iminuit
Fetching package metadata .........
Solving package specifications: .
> 
> UnsatisfiableError: The following specifications were found to be in conflict:
>   - iminuit -> python 2.7* -> openssl 1.0.1*
>   - python 3.6*

I'm not sure I understand the error message. openssl conflicts with python3?
@ParsonsRD, why are you using minuit and not `from scipy.optimize import minimize`?
@kosack, do we want minuit as a dependency in ctapipe?

tino-michael, 2017-05-18:
Ok, sorry, I just pulled from the master branch...
After running setup.py install again, minuit is working now.
Though the question remains: Why do we support a separate minimizer if there is already one in scipy?
Does minuit have some major advantages?

ParsonsRD, 2017-05-18:
Hi, typically I have found on complex multidimensional problems like this minuit just works better. There is a lot of code included in there for avoiding local minima etc. I have only made quick tests and found improved performance, but I agree we should think hard before release that we really need this as a dependency. I am hoping to get a student to check out the minimisers in the near future.

kosack, 2017-05-18:
Also, the new developer instructions for setting up ctapipe should install it automatically (the new ones have you run "conda env create -f environmnent.yml", which adds the necessary channel for iminuit (astropy), and installs it.

see here: https://cta-observatory.github.io/ctapipe/getting_started/

tino-michael, 2017-05-18:
I see, sorry for the buzz.
For my ANTARES reconstruction, I used TMinuit2 and just gave it several different starting positions for the fit to avoid local minima and they basically never ended up all in the same minimum. just a thought... Maybe you or your student can have a try at that too

closed 2017-05-18 by kosack