cta-observatory/ctapipe #419: dl1 calibrator does not apply gain-channel selection

kosack, 2017-05-24:
the DL1 image should be a single image, with no gain-channels left (in fact this may be done in R1→Dl0, but it needs to at least be in DL1). Therefore which gain channel used should be chosen when calculating the pixel intensity in `CameraDL1Calibrator`

@watsonjj do you have code to do the gain channel flattening or should we look into adding it (Tino has a basic implementation)

tino-michael, 2017-05-24:
here is my one-line implementation:
```
pmt_signal = event.dl1.tel[tel_id].image
pick = (pmt_signal>14) != np.array([True, False])[:, None]
picked = pmt_signal.T[pick.T] 
```

kosack, 2017-05-24:
this means we also need a way to specify the threshold  for the linear range of the high-gain amplifier (14 in Tino's example) for each camera that has 2 amps per pixel.    Should this be done in PE or in DC?

tino-michael, 2017-05-24:
I'd suggest in ADC counts, since the saturation itself should be independent of the calibration. Otherwise we would change the threshold for every pedestal/gain calibration, no?

kosack, 2017-05-24:
yes, I think DC is the correct thing to use, since that's the units in which the linearity is measured.  But in your example, it's using the PE, so we might want to do the gain switch at the DL0 level instead.

kosack, 2017-05-24:
>at the DL0 level instead.

in fact I think the Data Flow doc specifies that the cameras do the gain switch, so it would be in R1

watsonjj, 2017-05-24:
Okay, so the cameras provide a single channel, and we do not worry about it, except when deciding for MC?

watsonjj, 2017-05-24:
We will also need to consider which channel to use when reading the dst files, which are read directly into dl1.

watsonjj, 2017-05-24:
If this is done in units of DC, then this should be part of the MC r1 calibration (r0->r1, i.e. calib/camera/r1.py)

kosack, 2017-05-24:
> Okay, so the cameras provide a single channel, and we do not worry about it, except when deciding for MC?

yes, I think so, so we just need to implement a gain channel switch in the R1Calibrator... I do think we need to store somehow in DL0 which channel was used (as mentioned above) though, since to "recalibrate" that information would be needed, so a boolean mask would suffice for that.

for the DSTs, is it really not gain-selected already? I'm a bit surprised, since a DST is usually written after that step, but then yes I guess that logic would need to be there.  Perhaps that's the difference between "dst0" and "dst1"? I'll have to check the read_hess docs

watsonjj, 2017-05-24:
In the 1 dst file I have looked at, both channels were still obtainable.

kosack, 2017-05-24:
yeah, it looks like the DSTs are not so "summarized". Are the images in the dsts in PE or digital counts? If digital counts, we can just apply the same gain switch function

watsonjj, 2017-05-24:
DC, I apply the calibration into pe when reading them, so we can also have the gain switch when reading them

kosack, 2017-05-24:
Yes, that would work.  But then, what is the difference between the DSTs and raw data? Is it just the trace integration?

watsonjj, 2017-05-24:
Yes the integration is already performed. So the threshold for the gain switch would be different, correct?

watsonjj, 2017-05-24:
How would the gain switch be applied in the R1 calibration without integration? Or will you need to integrate the waveform?

kosack, 2017-05-24:
Hmm, yes the threshold might be: if any trace value exceeds a value, use the low gain, or it could be defined as an integral threshold... The latter is difficult if the window is not fixed, and the former is more robust. I wonder how it's done in read_hess?

What I suggest is that somebody (Tino or Jason perhaps?) adds this code to the R1Calibrator and any code needed to read DSTs (may need some thought). For the thresholds, we should create a that we can put into ctapipe_resources that can be loaded (later we will make it more flexible to be able to load from a db, or others). e.g.:

```python
from ctapipe.utils import datasets

threshtable = Table.read(datasets.get_dataset("gain_switch_thresholds.fits.gz"))
# the file is a table with 2 columns at least; cam_id and gain_switch_threshold
# or perhaps ecsv instead of FITS to make it easier to edit? (something readable by astropy.table is best

threshdict = dict(zip(threshtable['cam_id'], threshtable['gain_switch_threshold'])
```

then you can use `threshdict[geom.cam_id]` as the threshold

kosack, 2017-05-24:
>  Or will you need to integrate the waveform?

I know in VERITAS, the switch is done at the sample level. In HESS, we only read out fixed-window integrals, not sure about MAGIC?

kosack, 2017-05-24:
in read_hess, it seems to be done in the trace integration routine (controlled by the `sigamp` value), and seems to be at the sample level (though I didn't look in detail)

tino-michael, 2017-05-24:
FYI, my initial implementation breaks if the entry in the first channel is larger than the second -- which will happen anyway if we select on ACD. This one is more robust:
`pick = (pmt_signal > 14).any(axis=0) != np.array([True, False])[:, None]`

moralejo, 2017-05-25:
In MAGIC we have now just one gain. We only had a low gain branch in the
very first version of the MAGIC-1 camera. Back then, if I remember
correctly, the switch was done at the sample level.

-- 

Abelardo Moralejo Olaizola
Institut de Física d'Altes Energies
Tel : +34 931641662
Fax: +34 935811938
Avís - Aviso - Legal Notice - (LOPD) - http://legal.ifae.es

ParsonsRD, 2017-05-31:
Yes in HESS we make this decision based on ADC counts I believe.

closed 2019-08-05 by kosack