cta-observatory/ctapipe #445: Typed Items

MaxNoe, 2017-06-27:
I think it would be very usefull for i/o, provenance and sane defaults to introduce typed `Items` for containers.

Containers would then be very similiar to ORM Models, as for example in `peewee`. One could even imaging a direct translation for outputting containers into databases.

Possible non-exhaustive list:

* FloatItem
* IntegerItem
* MapItem
* ListItem
* ArrayItem
* StringItem
* DateTimeItem

This would make it clear, to which column types in FITS or HDF5 or ... one has to convert those entities.

What do you think?

kosack, 2017-06-27:
I'm happy with that. I originally considered it, but then just went with assuming the type of the default value (which of course may be confusing). This is basically the same design as any typical ORM, like SQLAlchemy.   It would be good to have the following metadata in each Item:
 
* type (by the item class)
* shape (all should support array operations, e.g. an IntItem could have a default shape of 1, a scalar, but in principle could be an array)
* unit 
* UCD (universal content descriptor, from the VO)
* description

I already support this in the HDFTableWriter, but it infers the type from the current value of the Container, not from the Items (which is error-prone), so this would be a better way to do it.

MaxNoe, 2017-09-06:
See #501

maxnoe, 2020-10-26:
@kosack implemented type checking for the fields in `validate` in #1163

closed 2020-10-26 by maxnoe