cta-observatory/ctapipe #448: metadata broken in Container

kosack, 2017-06-28:
I just noticed a problem in #442 (which is unfortunately already merged).   There is a misunderstanding with what the `.meta` attribute of containers was - it was *not* the Item metadata, but rather a set of header keywords, basically data that sould not become columns in an outtput file, but rather be headers.    Now it seems the item list (which was previously called "attributes") is mixed with "meta", which breaks writing HDF5 and FITS files..

Strangely the tests still pass, but just with warnings like `Keyword name 'tels_with_data' is greater than 8 characters or contains characters not allowed by the FITS standard; a HIERARCH card will be created.  keyword), VerifyWarning)`, and the output files are now full of crazy headers.

Can you change it back so that Container.meta is just a dictionary of headers?  Parts of the analysis also rely on that working.

Otherwise, I can revert the pull request.

kosack, 2017-06-28:
The old (correct) behavior was:

```python
c = Container(...)

c.meta['VERSION'] = 1.2  # becomes a FITS ,HDF, or other header keyword

c.attributes  # list of "Item"s in the container, used as column defs by `io.TableWriter`
```

These names are perhaps confusing, but the `meta` atrribute is the same as use in `astropy.table` for header metadata.  We can't call the other one `items`, since that has dict meaning, but it's tue that `attributes` is also confusing.  Maybe we need to rename the `Item` class to `Field`, and then have `attributes` renamed to `fields`?

kosack, 2017-06-28:
I'm just going to revert it for now, and add some test to make sure this doesn't fail, so you can simply re-open the original PR.

MaxNoe, 2017-06-28:
+1 for Field

kosack, 2017-06-28:
see #452  (quick refactor of Item→Field everywhere)

MaxNoe, 2017-06-28:
I already started with that :D

kosack, 2017-06-28:
Ok, I'll wait for your version :-)    I had to also modify one documentation page (I think io/index.rst) and examples/notebooks.  Otherwise, I just used the PyCharm refactoring tool to do it all automatically.

MaxNoe, 2017-06-28:
I do not really get the Idea of this `meta`. The writers all write a list of containers. 

But how should then `meta` be handled? How can it be header attributes if different containers have different values for the same keys?

kosack, 2017-06-28:
The writers don't really do what we need, which is why they are being phased out in favor of the TableWriter class which translates a Container as a row in an output table (which could be FITS, HDF5, text, a db, etc). The metadata is fixed information that *should not change* during writing (in fact the TableWriter only writes it on the first call to `write()`).    It is for things like the version number, array name, provenance info, etc, that are required in all output files in CTA. 

There may be other ways to handle this, of course (we could have a Container with the metadata that gets attached as headers via another writer), but a simple dict was  the simplest... We might consider something fancier in the future.

PR #453 (2017-06-28): Improve container
follow up to #442 fixes #448
closed 2017-06-28 by kosack