cta-observatory/ctapipe #466: MCHeaderContainer should be removed

kosack, 2017-07-10:
right now there is a `MCHeaderContainer` that contains only the `run_array_direction`.  There are a few problems with this:

1. `run_array_direction` should at least be a `astropy.coordinates.SkyCoord` or have units 
2. it means different thigns depending on the tracking mode, which is not recorded
3. this isnt' really monte-carlo specific information - we need the array tracking position in real data as well. 

Therefore this should be refactored so that we have a "pointing" container (`PointingContainer`) that can contain the necessary info, regardless if this is a MC or real data file. It also can be where pointing corrections are stored or applied.

kosack, 2017-10-06:
see #500

MaxNoe, 2020-03-20:
I think as this container now holds a plethora of monte carlo information, this is not longer a thing that should be removed

closed 2020-03-20 by MaxNoe