cta-observatory/ctapipe #471: common input for Reconstructors

kosack, 2017-07-18:
see discussion in the PR #458  , @tino-michael @ParsonsRD  

We need a common way of passing in the instrumental description (as a `SubarrayDescription` in the constructor) and the input parameters to the `predict()` method of all reconstruction routines.

tino-michael, 2017-07-18:
> tel_pointing is either a `Coordinate` or perhaps just a tuple of the theta,phi. I guess that is not per-telescope, but the array-wise pointing in the NominalFrame? or do you need different pointings for each telescope?

I need the pointing to transform positions on the cameras to directions in the **horizontal** frame, so I need it telescope-wise for now.

see the `guess_pix_direction` call in `get_great_circles`:
https://github.com/cta-observatory/ctapipe/blob/master/ctapipe/reco/HillasReconstructor.py#L264

ParsonsRD, 2017-07-19:
OK it depends what frame you are passing the data to the class, but for now I think it's best to have both.

mackaiver, 2018-12-11:
closed in light of #882

closed 2018-12-11 by mackaiver