cta-observatory/ctapipe #480: What are `pedestal file`, `transfer function file` and `adc2pe file` and what is the expected format

MaxNoe, 2017-07-25:
What are those? 

How are they used?

How can one produce them for existing experiments?

kosack, 2017-07-25:
I think these are defined only for CHEC right now (is that correct @watsonjj ?) They should at least be documented in the docs/calib directory (but are not right now, as far as I can tell)

To follow what we require for CTA, they should at least be FITS files (probably something readable by `astropy.table.Table` would be best, but the column definitions should be explicitly written down), but I have no idea what the current format is.

MaxNoe, 2017-07-25:
So why are those in a general CalibrationFactory if they are specific for CHEC?

watsonjj, 2017-07-25:
In the current implementation of factories, any traits that exist in one of the products of a factory need to also be included in the factory so that the trait value is propagated.

Those files are required in `TargetioR1Calibrator` which exists in an external module using ctapipe for analysis (for now). These file paths are supplied to TargetCalib, which is the calibration software for CHEC.

But CHEC is presumably not the only camera that will subtract pedestals, or convert from adc to pe in the r1 calibration. So these traits could be used by other cameras wishing to perform r1 calibration in ctapipe.

watsonjj, 2017-07-25:
Each camera team is responsible for their r1 calibration. What files they need, how they choose to apply the files, and even what format those files exist in depend on their choice.

watsonjj, 2017-09-01:
#489 Has included some extra documentation on these file paths, and R1 calibrators

watsonjj, 2018-01-31:
These traitlets are removed from the Factory now, as all traitlets are automatically obtained from the relevant Components and copied to the Factory as of #621 .

closed 2018-01-31 by watsonjj