cta-observatory/ctapipe #491: Add HESS telescope descriptions to Optics Configs

ParsonsRD, 2017-08-29:
As ctapipe reads the same format as the HESS simulations file I'd like to add the descriptions of the small and large telescopes to the camera and optics configs, that way reading them is fully supported. This would be nice for performance comparisons with existing software I think.

Does anyone think there are any other telescopes that should be added?

MaxNoe, 2017-08-29:
FACT could be added. Is there a format definition somewhere?

kosack, 2017-08-30:
That's fine with me at add any ones needed.  In fact, it is not specifically needed to add them to the FITS file (optics.fits.gz or *.camgeom.fits.gz)  since the optics definitions are generated on-the-fly from the monte-carlo files when they are loaded. The ones in the FITS files are just for testing purposes, or for working with cameras/telescopes when no monte-carlo file is opened (e.g to make toy-model simulations, or similar).

To add FACT and HESS to the optics descriptions, what really needs to be done is just to modify the lookup table at the top of `instrument/optics.py` and `instrument/camera.py` (I think HESS is already in the camera table, but not FACT, and neither are in the optics table so far)

MaxNoe, 2017-08-30:
FACT does not use simtel, so one cannot get it from mc files (Our mc files are fits files with the same structure as the data written by the telescope only containing some additional columns and headers with mc truth)

moralejo, 2017-08-31:
Dear Dan,

we would also like to have MAGIC, for the same reasons. Saša Mićanović,
from Rijeka, is implementing the MAGIC configuration for simulations with
simtelarray, so he will provide the needed data for ctapipe. He is now on
holidays but will be back on Monday.

Best regards,

Abelardo

-- 

Abelardo Moralejo Olaizola
Institut de Física d'Altes Energies
Tel : +34 931641662
Fax: +34 935811938
Avís - Aviso - Legal Notice - (LOPD) - http://legal.ifae.es

ParsonsRD, 2017-08-31:
OK, great well we can use this thread for collecting the required details then I'll make a pull request with them added. 

I think we just need the focal length, number of pixels in the camera. Then ctapipe should be able to guess the type.

kosack, 2017-08-31:
If you want to fill in all necessary info, you'll need to also have a custom event_source that fills the event.inst.subarray structure correctly (see the code in `hessio.py`).  As @ParsonsRD  said, you need those values and also the telescope ground positions (for reconstruction).  Then all you need to do is something like the following, replacing the pyhessio calls with whatever you use to lookup the information:

The `TelescopeDescription.guess()` method will use the lookup table in instrument/camera.py and optics.py to fill in the necessary info (though you can also do it manually)

``` python
          for tel_id in tel_ids:
                pix_pos = pyhessio.get_pixel_position(tel_id) * u.m
                foclen = pyhessio.get_optical_foclen(tel_id) * u.m
                mirror_area = pyhessio.get_mirror_area(tel_id) * u.m ** 2
                num_tiles = pyhessio.get_mirror_number(tel_id)
                tel_pos = pyhessio.get_telescope_position(tel_id) * u.m

                tel = TelescopeDescription.guess(*pix_pos, foclen)
                tel.optics.mirror_area = mirror_area
                tel.optics.num_mirror_tiles = num_tiles
                data.inst.subarray.tels[tel_id] = tel
                data.inst.subarray.positions[tel_id] = tel_pos
```

kosack, 2017-08-31:
An alternative to using the lookup tables (which are really just to get around the fact that the eventio MCs don't have the required info about what telescope types are being used), is to add  FITS files for the camera and edit the `optics.ecsv.txt` optics description in `ctapipe-extra/ctapipe_resources` (or put a copy somewhere in your `CTAPIPE_SVC_PATH`) and then construct the telescope using `TelescopeDescription.from_name("MAGIC","MAGICCam")` (or whatever you name them).

The only thing still missing in that scheme is the telescope ground positions - i still need to add a text or FITS file version of that.

I can help with all that if you give me the necessary info (pix pos in camera, focal length, num mirror facets, mirror area, array layout)

smicanov, 2017-09-06:
Hi guys,
I just joined the thread and catching up with what you need to add MAGIC into ctapipe.
Where to put or to whom to send the required info?
Cheers,
Sasa

labsaha, 2017-10-26:
Hi Karl & others,
I have the fits file containing MAGIC camera geometry. The data was provided by Sasa. I created the fits following the format used for other IACTs files. I wanted to test it locally if I can import the related information from that fits file through ctapipe. For that, I have added a line in the 'camera.py' script and copied the fits file to the 'ctapipe_resources' directory. However, I see that I cannot get the information about the MAGIC camera when I try the following command

```
from ctapipe.instrument import CameraGeometry
CameraGeometry.from_name('MAGICCam')
```
It is complaining that it cannot find the file
```
FileNotFoundError: couldn't locate table: MAGICCam.camgeom[.ecsv, .ecsv.txt, .fits, .fits.gz]
```
The name of the fits file is 'MAGICCam.camgeom.fits.gz' following the naming convention used for other IACT files. 
Also I added following lines in the 'camera.py' file, particularly at "_CAMERA_GEOMETRY_TABLE = {}
```
(1039, 17.5): ('LST', 'MAGICCam', 'hexagonal', 0 * u.degree, 19.1 * u.degree)
```
Do I need to do something more to make it work?

MaxNoe, 2017-10-26:
FACT: focal length 4.889 m, number of pixels 1440

kosack, 2017-10-27:
The `CameraGeometry.from_name()` method looks for the files in the ctapipe "service" path or in the ctapipe_resources module (which is inside the ctapipe-extra package).   So to find your new camera file, you just need to use the `CTAPIPE_SVC_PATH` environment variable to contain the directory where it is (this is a true *path*, not just  a single dir, so you can have multiple directories separate by ":" as with `$PATH`).  Then it should work. 

```
# assuming your camera file is in the current working dir:
export CTAPIPE_SVC_PATH=$CTAPIPE_SVC_PATH:$PWD
```
then

```
ipython -i --matplotlib=auto
>>> from ctapipe.instrument import CameraGeometry
>>> geom = CameraGeometry.from_name("MAGICCam")
```

The lookup-table in `camera.py` is just for creating cameras from e.g. MC files, where the name and type is not known, so they are constructed on-the-fly from metadata. 

If it all works, we can add it then to the `ctapipe-extra` package, along with the other files.

labsaha, 2017-10-30:
@kosack  Thanks a lot.  Now it's working. Just need to check few things before I send you that file so that you can add it to the ```ctapipe-extra```.

MaxNoe, 2017-10-31:
You could open a Pull Request yourself, no need to send files around ;-)

labsaha, 2017-11-20:
Hi,
Following the old simulation, the rotation in the configuration file for LST camera was 100.893 deg. This value was also considered for "LSTCam.camgeom.fits.gz" file provided in the ```ctapipe-extra``` package. Now the new configuration file for simulations of LST shows 10.893 deg rotation of the camera. Do we need another geometry file for LST camera in the ```ctapipe-extra``` package mentioning 10.893 in the header?

kosack, 2017-11-21:
> Do we need another geometry file for LST camera in the ctapipe-extra package mentioning 10.893 in the header?

It's probably not critical since so far we never use the files in ctapipe-extra during reconstruction, they are just there for testing purposes.  When you load up a MC file, it generates the cameras on-the-fly from the MC information, so the correct rotation should be there.   The reason for that is just that the MCs keep changing, so we can't rely on ctapipe-extra definitions being stable. Maybe in the future, that will be the case, but not as the MCs are being tweaked.

mackaiver, 2018-12-11:
FACTs is still in the wrong coordinate system. Its in the 'magic system' at the moment.

MaxNoe, 2020-03-20:
This got fixed a while ago

closed 2020-03-20 by MaxNoe