cta-observatory/ctapipe #492: Analysing FACT data with ctapipe

MaxNoe, 2017-08-31:
Here in Dortmund, we want to analyse FACT data with ctapipe. Measured or simulated does not really matter in the moment. Both have roughly the same format (gzipped FITS for simulations, zfits with an available python reader for observations).
This would make it possible to test ctapipe with real data and get a crosscheck for our own analysis software.

I was troubled on what exactly I need to implement or use to get the FACT data into a state where for example the hillas methods can be used.

Could someone guide me on what I need to implement? I started to do something here:
https://github.com/fact-project/ctapipe-fact
But couldn't really wrap my head around the different reader and calibration classes.

Could anyone guide me through the process? Any help would be greatly appreciated.

kosack, 2017-08-31:
There was a zfits event source, but I'm not sure if it's still maintained by anyone or still working - I suspect it needs some updating to get it running again since some of the data structures have changed, but that would be the first place to start. I've never tried it myself since I never had the necessary zfits library installed and had no test data to play with.   If it fills in the necessary info into the event Container just like `hessio_event_source`, the calibration and other code should work, as they are independent of the reader method.  

There is a separate wrapper class called EventFileReader that manages constructing various event sources, but you don't need to use that to start with (really we need to think about merging the two concepts, but that can be discussed later). 

For calibration, I believe most of what is there should work for FACT MCs, but I don't think all the code needed to calculate calibration coefficients from real data exists yet, though you might want to check with @watsonjj to see if you can adapt what he has created for the  CHEC camera.

watsonjj, 2017-08-31:
I'd be happy to help. Maybe to start I will paste the R1 calibrator we use for CHEC. Unfortunately we use svn instead of git for now.

watsonjj, 2017-08-31:
```
class TargetioR1Calibrator(CameraR1Calibrator):
    name = 'TargetioR1Calibrator'
    origin = 'targetio'

    pedestal_path = Unicode('', allow_none=True,
                            help='Path to the TargetCalib pedestal '
                                 'file').tag(config=True)
    tf_path = Unicode('', allow_none=True,
                      help='Path to the TargetCalib Transfer Function '
                           'file').tag(config=True)
    pe_path = Unicode('', allow_none=True,
                      help='Path to the TargetCalib pe conversion '
                           'file').tag(config=True)
    ff_path = Unicode('', allow_none=True,
                          help='Path to a TargetCalib flat field '
                               'file').tag(config=True)

    def __init__(self, config, tool, **kwargs):
        """
        Parent class for the r1 calibrators. Fills the r1 container.

        Parameters
        ----------
        config : traitlets.loader.Config
            Configuration specified by config file or cmdline arguments.
            Used to set traitlet values.
            Set to None if no configuration to pass.
        tool : ctapipe.core.Tool
            Tool executable that is calling this component.
            Passes the correct logger to the component.
            Set to None if no Tool to pass.
        kwargs
        """
        super().__init__(config=config, tool=tool, **kwargs)

        self.calibrator = None
        self.telid = 0

        self._load_calib()

    def calibrate(self, event):
        """
        Fake function to satisfy abstract parent, this is overloaded by
        either fake_calibrate or real_calibrate.
        """
        pass

    def _load_calib(self):
        if self.pedestal_path:
            self.calibrator = target_calib.Calibrator(self.pedestal_path,
                                                      self.tf_path,
                                                      [self.pe_path,
                                                       self.ff_path])
            self.calibrate = self.real_calibrate
        else:
            self.log.warning("No pedestal path supplied, "
                             "r1 samples will equal r0 samples.")
            self.calibrate = self.fake_calibrate

    @observe('pedestal_path')
    def on_input_path_changed(self, change):
        new = change['new']
        try:
            self.log.warning("Change: pedestal_path={}".format(new))
            self._load_calib()
        except AttributeError:
            pass

    @observe('tf_path')
    def on_tf_path_changed(self, change):
        new = change['new']
        try:
            self.log.warning("Change: tf_path={}".format(new))
            self._load_calib()
        except AttributeError:
            pass

    def fake_calibrate(self, event):
        if event.meta['origin'] != self.origin:
            raise ValueError('Using TargetioR1Calibrator to calibrate a '
                             'non-targetio event.')

        if self.check_r0_exists(event, self.telid):
            samples = event.r0.tel[self.telid].adc_samples
            event.r1.tel[self.telid].pe_samples = samples

    def real_calibrate(self, event):
        if event.meta['origin'] != self.origin:
            raise ValueError('Using TargetioR1Calibrator to calibrate a '
                             'non-targetio event.')

        if self.check_r0_exists(event, self.telid):
            samples = event.r0.tel[self.telid].adc_samples[0]
            fci = event.r0.tel[self.telid].first_cell_ids
            r1 = event.r1.tel[self.telid].pe_samples[0]
            self.calibrator.ApplyEvent(samples, fci, r1)
```

watsonjj, 2017-08-31:
Our calibration works using an external library, TargetCalib, which we supply it paths (pedestal, tf, adc2pe/pe). If no pedestal path is supplied, no calibration makes any sense, so `fake_calibrate` will be used and the r1 waveforms will equal the r0 waveforms. If a pedestal path is supplied then `real_calibrate` is called and TargetCalib calibrates the waveforms.

If you could give me some information of what calibration you want to do on the waveforms, I can help further.

MaxNoe, 2017-09-05:
We have the necessary calibration routines to calibrate the drs4 data to calibrated voltages, 
implemented here:
https://github.com/fact-project/zfits

What are the expected amplitudes of R1? Units of a single photon pulse?

watsonjj, 2017-09-05:
I believe so. The units we use for CHEC are such that the dl1 value (integrated charge) is 1 for a single photon pulse.

MaxNoe, 2017-09-05:
But this will be different for the different extraction methods, or am I mistaken?

watsonjj, 2017-09-05:
Yes, one peak finding method could give you a different charge value to another.

You may also be concerned about the dependance on integration window length, but this is handled by the integration correction in dl1.py.

Our approach in CHEC was to use AverageWfPeakIntegrator on non-p.e.-calibrated waveforms with a single p.e. illumination. We used this Integrator because we illuminated all pixels, and all the pulses occur at the same time in our waveform, therefore getting a very good handle on peak position. From this we constructed a spectrum of integrated charges and fit it to find the amount of charge that a single photoelectron results in (per pixel). Then for future measurements we multiply each pixel by the inverse of that value.

As a result of this approach, as long as the peak finding method correctly finds the peak on the p.e. calibrated waveforms, then the resulting charge should be the correct amount in p.e.. Of course no peak finder is perfect, especially with Cherenkov showers, which leads to the different charge resolutions between peak finding methods.

MaxNoe, 2017-09-05:
Ok, this is the same how it is done in FACT.

MaxNoe, 2017-09-05:
I'm now trying to use the `LocalPeakIntegrator` together with the `CameraDL1Calibrator`.

I have a few questions:

1. the `PeakFindingIntegrator` has traits `sig_amp_cut_HG` and `sig_amp_cut_LG`. How should I set these for a Camera with only one channel.

2. When I try to apply the calibration, I get:
```
  File "/home/maxnoe/.local/anaconda3/lib/python3.6/site-packages/ctapipe/calib/camera/dl1.py", line 240, in calibrate
    correction = self.get_correction(event, telid)[:, None]
  File "/home/maxnoe/.local/anaconda3/lib/python3.6/site-packages/ctapipe/calib/camera/dl1.py", line 195, in get_correction
    step = event.mc.tel[telid].meta['refstep']
KeyError: 'refstep'
```

What is refstep? Why is it required but in `meta`? Shouldn't required objects be not in `meta` but a required member of the fitting container?

watsonjj, 2017-09-05:
1. You do not need to set these, I do not. But if you wanted to only consider values in your waveform above a certain value, then set sig_amp_cut_HG, which is applied to the first channel.

2. To apply the integration correction you need to set the reference pulse shape parameters. These are `event.mc.tel[telid].reference_pulse_shape`, `event.mc.tel[telid].meta['refstep']` and `event.mc.tel[telid].time_slice`. This information is automatically filled from simtelarray files, so what I did was to copy and hard code the MC values for the CHEC camera into the reader for CHEC data.
This is something that should be moved to the instrument module in the future.

MaxNoe, 2017-09-05:
1. Ok thanks

2. I found that the wrong or to less types of exceptions was catched if the mc information is not there:
https://github.com/cta-observatory/ctapipe/pull/496

closed 2018-12-11 by mackaiver