cta-observatory/ctapipe #499: Pointing information

MaxNoe, 2017-09-05:
There does not seem to be a container for the array pointing information.

Should I add one?

kosack, 2017-09-05:
Yes, probably.  Although where does the current reconstruction get it? I guess from the MC container, which is not sufficient.  At least we need the corrected alt and az in a container (since that is what will be written, and eventually converted to RA/Dec).  It could be separate fields or stored as an `astropy.coordinate.SkyCoord` in the `AltAz` frame (I guess if we are using containers to serialize to disk, separate alt/az values may be better, though)

kosack, 2017-09-05:
There also needs to be a way to get each telescope's tracking position, since that will be recorded and used in the reconstruction (the nominal center point is less useful).

kosack, 2017-09-05:
@tino-michael how does your reconstruction get this info currently?

MaxNoe, 2017-09-05:
> There also needs to be a way to get each telescope's tracking position, since that will be recorded and used in the reconstruction (the nominal center point is less useful). 

This is what I was talking about

tino-michael, 2017-09-05:
currently, this information is accessible via `event.mc`:
```
tel_phi[tel_id] = event.mc.tel[tel_id].azimuth_raw * u.rad
tel_theta[tel_id] = (np.pi/2-event.mc.tel[tel_id].altitude_raw)*u.rad
```

kosack, 2017-09-06:
It's probably best to have a container for this per telescope, something like `TelescopeTrackingContainer`, since there may be more info than just the az/alt to be added later (like velocities, current tracking error, etc).  That way we can also read/write it as a table. You can think of Containers as basically database or FITS/HDF5 table schemas. 

Then we can add a map of those to the R0Container (or DL0?)

kosack, 2017-09-06:
or it can be a `TrackingContainer` directly under the DataContainer, like the `CentralTriggerContainer`, and contain a map of telescope-wise `TelescopeTrackingContainers`.  Probably we need to go through all the container hierarchy at the next workshop and decide how to make it more consistent...

MaxNoe, 2017-09-06:
I would go for a `TrackingContainer` in the `DataContainer` which should probably renamed to `ArrayEventContainer` to be more specific?

kosack, 2017-09-13:
see #504

closed 2019-01-03 by MaxNoe