cta-observatory/ctapipe #500: Clean up the Container hierarchy

kosack, 2017-09-06:
We need to look in detail at all the data that is currently in the Container hierarchy, as well as data that needs to be added (for things like Tracking, Pointing, HV monitoring, etc.) and decide if:

* the structure is consistent or needs updating
* the use-cases (e.g. how data are read or written into them) are met or understood
* data items are missing or redundant

Note that the idea for at least some Containers is that they are schemas for output or input files (and can thus be serialized to HDF5, FITS, or database tables).  So making them fit into that context is important.

MaxNoe, 2017-09-06:
See also my proposal for typed fields: #501

kosack, 2017-09-06:
Yes, perhaps we should wait until the typed fields are implemented, and then we can re-evaluate the contents.

kosack, 2017-10-06:
It was decided in Annecy that we will change the container hierarchy such that the *event* containers only contain things that change per-event, and there will be a separate *header* (better name hopefully) container that has the instrument and other observation-specific information inside it.

closed 2026-11-10 by maxnoe