cta-observatory/ctapipe #505: Why do CameraFrame and HorizonFrame need an `ArrayPointing`

MaxNoe, 2017-09-08:
Transformation from Camera Coordinates to sky coordinates should only require the pointing of the single telescopes, or not?

kosack, 2017-09-08:
I think the coordinate transforms are a bit backward and need reworking (see e.g. #416)

ParsonsRD, 2017-11-21:
Short answer, they don't. However if they are defined here then a direct transformation can be performed all they way through all the systems. e.g. from camera directly to Horizon.

Agreed this needs restructuring though.

MaxNoe, 2017-11-21:
Why does a transform from `CameraFrame` to `AltAz` need an `ArrayPointing`? 

The individual pointing of the Telescope should be enough.

ParsonsRD, 2017-11-21:
Yes, it doesn't but some of the frames in between do. Currently the coordinates transform doesn't bypass any systems, such as the system centred on the nominal array pointing direction, but that is easy to change in the astropy structure.

MaxNoe, 2018-12-19:
fixed in #896

thomasgas, 2019-01-31:
Solved in #896

closed 2019-01-31 by thomasgas