cta-observatory/ctapipe #514: FlashCam digital filter

GernotMaier, 2017-09-26:
FlashCam is using a digital filter algorithm which is currently not applied in the simulation. This should be implemented in a step before applying the trace analysis

Attached a c-code and documentation provided by the FlashCam team.

Quote from the original message:

> The MC-guys should use

> "PzpsaSmoothUpsample"
> or
> "PzpsaSmoothUpsampleFloat"
> depending on  input data type.
> 
> The upsample value (us) must be "4", i.e. to upsample into GHz.
> 
> They need to know (provide as input) the baseline of the signal in digital units;
> The pole zero (pz) value can be determined with
> "PzpsaGetDecay"
> by applying it to the input pulse shape.
> The resulting pz value should be around 0.75 . The deconvolved pulse will be ~gaussian with a FWHM of  ~ 8 nsec (verified: 7.9 sec in the real camera protoype data).
> 
> The devonvolved shape is then NOT integrated anymore, but rather evaluated at a certain point in time (typically certainly at it's maximum position). For higher-amplitude signals, or broader signals the signal can be integrated over a range to reduce phase errors. Certainly the proper, relative calibration-factor for different integration times must be used.
> The routine
> "PzpsaPeakProperty"
> does the job, including cog calculation.
> 

Attached are (as a tar package):

- documentation (pzpsa.html)
- original code from FlashCam team (pzpsa.h and pzpsa.c)
- pulse_FlashCam_raw2.dat (prod3 FlashCam pulse shape)
- pulse_SST1M_AfterPreamp.dat (prod3 SST-1M pulse shape)
- poleZero.C small root (!!!) macro to calculate the pole zero value from a given pulse shape (0.75 is a good value for the prod3 FlashCam pulse shape)
- two plots showing an example of how the pulse shapes look before and after the digital filtering 

Note: the first and last 3-10 samples need to be excluded from the trace integration of the filtered trace (as there are edge effects)

[digitalFilter.tar.gz](https://github.com/cta-observatory/ctapipe/files/1332842/digitalFilter.tar.gz)
[image]
[image]

watsonjj, 2017-09-26:
Great. In GCT we have also began to use a cross-correlation with the known pulse shape for low amplitude pulses, and I have been thinking how best to handle this in ctapipe.

Currently there exists a image.waveform_cleaning module, which contains algorithms that are applied in the DL1 calibration before the charge extraction methods are used (contained in image.charge_extractors). Therefore I would create the digital filter in the image.waveform_cleaning (as a c extension if it cannot be efficiently created using numpy), and then a charge extractor that follows the method described (if there is not a charge extractor that already does exactly what is needed).

There is one update to the current situation in ctapipe that would be good. Some cameras are going to specify the **exact** algorithms that should be applied to their waveforms, e.g. Flashcam. There should be a module or method that initialises the correct methods based on the camera being investigated...I am currently thinking about the best way to do that, and I would be open to suggestions.

watsonjj, 2017-09-26:
Alternatively the entire algorithm could be implemented in a single "FlashcamChargeExtractor"... I'm not sure which is the best way...

kosack, 2017-10-05:
The question also is if we use *Exactly* their code (or a copy of it), or implement the algorithm as it is described, e.g. using the filtering functions of `scipy.signal`. I suspect we could boil it down to a few lines of code in the second case.

maxnoe, 2020-10-26:
Is this still an issue?

watsonjj, 2020-10-26:
Yes, it has not been implemented yet

maxnoe, 2021-03-19:
This requires making it possible to use an image extractor as a telescope parameter.

maxnoe, 2021-12-06:
> Make ImageExtractor a TelescopeParameter #1660

This would now be possible to implement. Anyone from the FlashCam team willing to contribute an extractor for flashcam?

kosack, 2021-12-08:
I don't think anyone from the FlashCam teams is on this GitHub.  We need this to properly analyze Prod-5 CTA-South, so it's starting to become more important.  The main issue again is  that we have their C software for their digital filter, however the algorithm is not documented as far as I know, or at least I couldn't find documentation.  There are API docs, but even those are sparse, and there are many functions that are not actually used.

Ideally, we would find somebody who could properly explain the filter algorithm, and then we can implement it in SciPy. The fall back would be to include their c-code as a library, but that is less nice since we have to re-compile it for every architecture, and I dislike undocumented algorithms being included.

closed 2023-03-30 by maxnoe

maxnoe, 2023-03-31:
After five and a half years, this issue was finally closed thanks to @clara-escanuela and @fwerner in #2188