cta-observatory/ctapipe #515: Effective focal length

GernotMaier, 2017-09-26:
Effective focal length differs from nominal focal lengths - see attached table by Konrad (calculated based on the prod3 telescope model)

This is especially important for the direction reconstruction for off-axis gamma rays. The plate scale of each telescope should be scaled by these values.

[flen.pdf](https://github.com/cta-observatory/ctapipe/files/1332851/flen.pdf)

kosack, 2017-10-05:
What we call "effective_focal_length" currently is the "right" one to use for reconstruction (not the same definition as Konrad uses), so perhaps it should be named more clearly. However, it would be nice to have both somehow read out.

kosack, 2017-10-06:
I think right now there is no way to read the nominal or effective focal lengths from the MC files.  

Would it suffice for us to rename `effective_focal_length` to `equivalent_focal_length`? That would avoid confusion between definitions (the one we use now is the equivalent one, assuming a single-mirror, so it works for both single and dual mirror systems).

PR #575 (2017-11-09): Refactor optics focal length name
- changed `effective_focal_length` to `equivalent_focal_length`,  fixes #515 
- added `.info()` functions to `OpticsDescription` and `CameraGeometry` for convenience
- added a better test for camera pixel neighbors (testing all cameras)

This change implies also a change to the `optics.ecsv.txt` table in *ctapipe-extra*, which has been updated (though the old format where the column name is still "effective" is still readable, with a warning emitted)
closed 2017-11-09 by kosack