cta-observatory/ctapipe #516: Check Calibration Example

Sojourner92, 2017-09-26:
In /Examples/Notebooks/check_calib.ipynb, I get an index error with the following traceback:

IndexError                                Traceback (most recent call last)
<ipython-input-4-3a15d4209c96> in <module>()
     10 for event in tqdm_notebook(hessio_event_source(infile, max_events=max_events), total=max_events):
     11     cal.calibrate(event)
---> 12     process_pedestal(event)

<ipython-input-3-1215205017ef> in process_pedestal(event, output)
     16                           im[~mask].mean(), im[~mask].std()))
     17 
---> 18         hist, ed = np.histogram(im[~mask], bins=200, range=[-10,10])
     19 
     20         if geom.cam_id in hists:

IndexError: boolean index did not match indexed array along dimension 1; dimension is 1 but corresponding boolean dimension is 2368

kosack, 2017-09-27:
this seems to be a problem introduced by the recent calibration update when encountering an ASTRI camera.  @watsonjj do you know if this is normal? The problem is that the image has a strange number of dimensions that doesn't match the other cameras.

kosack, 2017-09-27:
It seems that for astri telescopes (with no traces), we end up with a DL1 image that has an extra dimension (of length 1) that causes some problems.  See below (tel 34 is ASTRICam):

```
ipdb> event.dl1.tel[13].image.shape
(1, 1764)
ipdb> event.dl1.tel[25].image.shape
(1, 1764)
ipdb> event.dl1.tel[34].image.shape
(2, 2368, 1)
```

kosack, 2017-09-27:
adding `np.squeeze(im)` in the example fixes it, but something should be fixed in `dl1.py` most likely so that the outputs are all the same for all telescopes.

kosack, 2017-09-27:
I opened #520 to fix this generally, but will just update the example for now (it will likely cause problems in other parts of the code, however, so that issue needs fixing)

commit 91d22e6c (2017-09-27): Merge pull request #522 from kosack/fix_check_calib_example

updated notebook to fix bug, fixes #516
closed 2017-09-27 by kosack