cta-observatory/ctapipe #556: Write special DL1 data for deep-learning studies

kosack, 2017-10-06:
A use case from Daniel Nieto:

Let me try to describe one of our own use cases, where I tried to use
current ctapipe functionalities as far as my limited knowledge of the
code allows.

1. A set of simtel files are read and processed with ctapipe, stopping
at DL1. For each event and triggered telescope, the integrated charge
("image" in the DL1 container) and arrival time ("peakpos" in the DL1
container) for each pixel are now accessible as two 1-D arrays whose
length is the number of pixels in the camera (telescope-type dependent).

2. Both 1-D arrays from the container above (charge and timing) are
transformed into a 3-D arrays representing the camera geometry. I
believe this can be already done with
ctapipe.image.convert_geometry_hex1d_to_rect2d - if a rotation of 0 deg
is applied, I guess - , but not for the square-pixel cameras, right? The
resulting 3-D array (rot_img if convert_geometry_hex1d_to_rect2d is
used) is stored to disk in hdf5 format, along with all the mc, trigger
and instrumental info.

I would say the output from points 1 and 2 could be the common format
those of us working with machine learning methods would love to have
(this is the impression I got from our last two dedicated calls). From
now on the use case describes what we would do with those data.

3. DL1 data in h5 format is input to our deep learning models. Our deep
learning software stack is based on the Tensorflow python API. Our h5 to
tensorflow interface could then select which events to train/eval our
models with, e.g., cutting in multiplicity, telescope type, or any
combination of parameters from the h5 metadata.

I hope all this makes sense...

If our deep learning studies (or any of the ongoing alternative event
characterization studies) prove to be comparable or outperform the
standard event characterization we would work on their implementation
within the ctapipe.reco module (at least that would me my personal
intention).

kosack, 2017-10-06:
I think this is partially supported by PR #554 , but would need a new `Container` class to store the *2D cartesian* images and a script to do the geometry conversion before writing out (using `image.GeometryConverter`).

kosack, 2017-10-12:
@ParsonsRD is there a particular data format you are using for your CNN studies?

ParsonsRD, 2017-10-12:
Just reading from Konrad's day level 0 data at the moment.

ParsonsRD, 2017-10-12:
DST level 0 that is

nietootein, 2017-10-25:
Hi all,

We are starting to compile a list of quantities we'd like to have stored to disk for our ML studies that can be found [here](https://docs.google.com/document/d/16NSEclzn2cPRGeqPTMmRO4Lt1yq4kOqSGqVkv8QE_DA/edit?usp=sharing).

Some of this quantities are run-wise, some others are even-wise, and not all of them come from the DL1 container (e.g. mc info, instrument data, ...).  I do not know if the scope of the DL1 dumper (PR #554 ) feature also includes adding variables from other containers, but that would be very nice. 

Regarding the storage of _2D cartesian_ images: for the sake of disk space, we would like to investigate if performing the 1D to 2D conversion on-the-go while we are loading the datasets for training does not have a significant impact on performance. 

Cheers,

D.

Sojourner92, 2017-11-12:
I have a script that makes square integrated images/ datacubes with waveforms for each pixel included for the CHEC cameras (and saving them to hdf5) that I could submit to ctapipe. But we need to discuss on the next phone call the best way to train the networks on the waveforms before that. I would personally advise against using the 1D to 2D conversion when training; I'd say its better just to do it once when extracting the data from the simtel files using ctapipe and then save to hdf5. That way you can train as many networks as you like based on your dataset as opposed to having to perform the image squaring every time. 

Could anyone please elaborate on the differences between dl1 and r0 data for MC events as opposed to real data? Myself and Etienne Lyard discussed this in La Palma and we're both using the r0 level data in order to avoid the calibration biasing the training.

watsonjj, 2017-11-12:
For MC files:
R0 - Raw
R1 - Pedestal subtracted and samples converted to p.e.. Both the pedestal and dc_to_pe arrays are also extracted from the sim_telarray file. There is no special analysis done for this calibration.
DL0 - Same as R1
DL1 - Charge extracted from the waveform per pixel (1 value per pixel).

If you are training your algorithms using waveforms I would suggest to use DL0. The waveforms at this level should match closest to the real data, and this is the earliest data level that would be stored in the file array implementation for use in offline analysis.

vuillaut, 2017-11-12:
Hi Sam.

I wish you were there the evening we discussed these points during the
consortium meeting.
There are already at least two scripts available in ctapipe that allow you
to shift any camera into a square matrice, one of which conserves data
locality in memory in respect to pixels positions (written by Pierre
Aubert).

Now concerning 2D conversion, we agreed not to it for storage - but as you
were not there we can of course discuss it during next phone call. The main
reason is that 2D conversion to a square matrice introduce a geometrical
bias that will depend on the algorithm you use. If we want to put things in
common, I'd advise to keep things as general as possible and not make
choices that would impact everybody. Of course, I agree there is a cost to
that, and it is an extra computation at training. But once we have a common
dataset, nothing prevents you from doing it on your own machine and storing
it locally.

Concerning waveforms and calibration. I see Jason replied concerning the
data levels. Unless you are plugged online, you will never access R0 and R1
data. I'm also thinking about how we can use deep learning to select events
on site without calibration but this is a very long shot and we are very
far from that. You'll run into huge problems as two images won't be
comparable at all, the calibration evolving quickly in time during
observations (not the case for simulated data).
Here again, we agreed to start the work with calibrated and integrated (no
waveforms) data during the meeting. Also can be discussed.
The main reasons were to start "simple" before seeing too big and to have
manageable data volumes as the whole production once calibrated and
integrated boils down to several TB compared to several hundreds of TB for
waveforms.

Sojourner92, 2017-11-12:
Hi all,
Thanks for the answer Jason and the response Thomas; I also regret not being at there that evening and agree that we should discuss all the above during the phone call tomorrow (what time is it by the way, the call is not on the ASWG indico or the machine learning redmine?).  One of my main interests over the past few weeks has been considering using machine learning techniques to quantify the benefits of taking waveform data in the cameras, however I'd only plan to be using a comparitively small dataset for this. I agree with all your points for larger runs.

The one point I would raise with respect to ctapipe is that if there is a script to convert the more rectangular cameras such as CHEC to square matricies as you say, then it needs to be documented; at the minute there's only reference to a script for converting hexagonal camera output to square images.

nietootein, 2017-11-13:
Hi all,

The next ML call would be next [Monday 20th, at 10:00 CET](https://indico.cta-observatory.org/event/1645) (we use the aswg mailing list for announcements). We will be summarizing and following up on the discussions that took place in La Palma, so everyone should be up to speed after the call. Sorry if I missed some of you when organizing the informal meeting in La Palma! It seems I have not identified all ML fans around...

Cheers,

Daniel

kosack, 2017-11-15:
> r0 level data in order to avoid the calibration biasing the training.

That seems like a bad idea - with uncalibrated cameras, you will train on the calibration artifacts, not the  images. The DL0 images have e.g gain variations and noise pedestal variations, and in reality those will be much more complex and may change in time, and therefore break your model on real data.  The idea of calibration is to remove non-physical artifacts from the data so that they can be analyzed.  Any machine learning should start at the DL1 level.

watsonjj, 2017-11-15:
> DL0 images have e.g gain variations and noise pedestal variations

Aren't those variations corrected for by the DL0 level? 

If you want to create a machine learning algorithm (or any other algorithm) that utilises the entire waveforms as opposed to extracted charge images, isn't DL0 your best (and really your only) option?

Granted it is a lot more complex and more likely to break on real data...

Sojourner92, 2017-11-17:
So, I'm using ctapipe's ctapipe.io.hessio.hessio_event_source reader to read in data from a simtelarray file, below is what I get if I print out an event. It seems to me that if I want to use the MC waveform data then I don't really have much of a choice but to use r0 at the minute, or did I miss some DL0/DL1 flag when I ran simtelarray? Granted I'm the only person currently interested in using the waveform data for the machine learning.
```
{'count': 0,
 'dl0': {'event_id': 300,
         'run_id': 1,
         'tel': {},
         'tels_with_data': {1, 2, 3, 4}},
 'dl1': {'tel': {}},
 'dl2': {'classification': {}, 'energy': {}, 'shower': {}},
 'inst': {'mirror_dish_area': {1: <Quantity 12.566370964050293 m2>,
                               2: <Quantity 12.566370964050293 m2>,
                               3: <Quantity 12.566370964050293 m2>,
                               4: <Quantity 12.566370964050293 m2>},
          'mirror_numtiles': {1: 2, 2: 2, 3: 2, 4: 2},
          'num_channels': {1: 1, 2: 1, 3: 1, 4: 1},
          'num_pixels': {1: 2048, 2: 2048, 3: 2048, 4: 2048},
          'optical_foclen': {1: <Quantity 2.2829999923706055 m>,
                             2: <Quantity 2.2829999923706055 m>,
                             3: <Quantity 2.2829999923706055 m>,
                             4: <Quantity 2.2829999923706055 m>},
          'pixel_pos': {1: <Quantity [[-0.1022    ,-0.0958    ,-0.0894    ,...,  0.0894    ,
             0.0958    , 0.1022    ],
           [-0.15539999,-0.15539999,-0.15539999,...,  0.15539999,
             0.15539999, 0.15539999]] m>,
                        2: <Quantity [[-0.1022    ,-0.0958    ,-0.0894    ,...,  0.0894    ,
             0.0958    , 0.1022    ],
           [-0.15539999,-0.15539999,-0.15539999,...,  0.15539999,
             0.15539999, 0.15539999]] m>,
                        3: <Quantity [[-0.1022    ,-0.0958    ,-0.0894    ,...,  0.0894    ,
             0.0958    , 0.1022    ],
           [-0.15539999,-0.15539999,-0.15539999,...,  0.15539999,
             0.15539999, 0.15539999]] m>,
                        4: <Quantity [[-0.1022    ,-0.0958    ,-0.0894    ,...,  0.0894    ,
             0.0958    , 0.1022    ],
           [-0.15539999,-0.15539999,-0.15539999,...,  0.15539999,
             0.15539999, 0.15539999]] m>},
          'subarray': SubarrayDescription(name='MonteCarloArray', num_tels=4),
          'tel_pos': {1: <Quantity [ 80., 80.,  5.] m>,
                      2: <Quantity [-80., 80.,  5.] m>,
                      3: <Quantity [ 80.,-80.,  5.] m>,
                      4: <Quantity [-80.,-80.,  5.] m>},
          'telescope_ids': [1, 2, 3, 4]},
 'mc': {'alt': <Angle 1.2217304706573486 rad>,
        'az': <Angle 1.570796251296997 rad>,
        'core_x': <Quantity -0.0 m>,
        'core_y': <Quantity -0.0 m>,
        'energy': <Quantity 1.3132532835006714 TeV>,
        'h_first_int': <Quantity 27633.1484375 m>,
        'tel': {1: {'altitude_cor': 0.0,
                    'altitude_raw': 1.2217304706573486,
                    'azimuth_cor': 0.0,
                    'azimuth_raw': 1.5707963705062866,
                    'dc_to_pe': array([[ 0.01446366,  0.015482  ,  0.01526728, ...,  0.01483649,
         0.01579109,  0.0146652 ]]),
                    'pedestal': array([[ 3819.19189453,  3881.48291016,  3721.17871094, ...,
         3773.23388672,  3783.74975586,  3745.4284668 ]]),
                    'photo_electron_image': array([0, 0, 0, ..., 0, 0, 0], dtype=int32),
                    'reference_pulse_shape': array([[  3.18908691e-02,   3.53698730e-02,   3.88183594e-02,
          4.22973633e-02,   4.57458496e-02,   4.99572754e-02,
          5.49011230e-02,   5.98449707e-02,   6.48193359e-02,
          6.97631836e-02,   7.56225586e-02,   8.24584961e-02,
          8.92333984e-02,   9.60693359e-02,   1.02905273e-01,
          1.10778809e-01,   1.19812012e-01,   1.28784180e-01,
          1.37817383e-01,   1.46850586e-01,   1.57104492e-01,
          1.68579102e-01,   1.80053711e-01,   1.91406250e-01,
          2.02880859e-01,   2.15698242e-01,   2.29614258e-01,
          2.43652344e-01,   2.57568359e-01,   2.71484375e-01,
          2.86621094e-01,   3.02978516e-01,   3.19335938e-01,
          3.35937500e-01,   3.52294922e-01,   3.69628906e-01,
          3.87939453e-01,   4.06250000e-01,   4.24560547e-01,
[... 377 lines cut ...]
        [41, 39, 46, ..., 43, 44, 46],
        [46, 45, 51, ..., 41, 39, 38],
        ..., 
        [36, 35, 31, ..., 33, 36, 40],
        [42, 36, 39, ..., 36, 37, 38],
        [36, 40, 45, ..., 37, 40, 39]]], dtype=uint16),
                    'adc_sums': array([[3617, 3969, 3795, ..., 3373, 3694, 3701]], dtype=int32),
                    'num_samples': 96},
                4: {'adc_samples': array([[[37, 37, 35, ..., 42, 39, 39],
        [36, 36, 39, ..., 37, 37, 37],
        [46, 47, 45, ..., 39, 38, 39],
        ..., 
        [35, 37, 39, ..., 38, 41, 39],
        [36, 37, 34, ..., 41, 39, 39],
        [39, 35, 40, ..., 45, 41, 44]]], dtype=uint16),
                    'adc_sums': array([[3965, 4074, 4195, ..., 3920, 3922, 3778]], dtype=int32),
                    'num_samples': 96}},
        'tels_with_data': {1, 2, 3, 4}},
 'r1': {'event_id': 300,
        'run_id': 1,
        'tel': {},
        'tels_with_data': {1, 2, 3, 4}},
 'trig': {'gps_time': <Time object: scale='utc' format='unix' value=1508747702.180526>,
          'tels_with_trigger': array([1, 2, 3, 4], dtype=int32)}}
```

watsonjj, 2017-11-17:
The DL0 and DL1 containers are filled after performing the DL0 and DL1 calibration. Take a look at ctapipe/examples/calibration_pipeline.py. I can explain in more detail in person next week.

closed 2018-04-25 by kosack