cta-observatory/ctapipe #558: clean up tests that load data

kosack, 2017-10-06:
right now a lot of tests load up events from test data files, which uses a lot of memory and time.  This could be improved by defining a single pytest *fixture* that loads an event (or 2 or 3) once, and that can be reused in any test that needs an event. 

A second problem is that the current test file gamma_test.simtel.gz is very out-of-date (prod2, with a super-array), and thus doesn't cover all cameras correctly.  Additionally, if we recompile pyhessio for a smaller memory footprint, it will no longer be readable.

So I would suggest two improvements:
 - add a test fixture for event-loading
 - change `gamma_test.simtel.gz` to a more modern file that better represents the latest CTA (perhaps we need 2: one for north and one for south?)
 - update the calibration tests (which seem to look for specific results from this particular file) to either use a new file, or to use dummy data that is not read from a file.

kosack, 2020-05-15:
This is really 2 issues.  The first is fixed (only one event is loaded, as a fixture).

maxnoe, 2022-10-27:
The second part is fixed as well (we now use `gamma_prod5.simtel.zst` for most tests.

I also quickly had a look at the calibration tests, and they are sing mostly random numbers, not the actual waveform data of the example event.

So I'll close this.

closed 2022-10-27 by maxnoe