cta-observatory/ctapipe #560: Implement IRF output in required format

kosack, 2017-10-10:
It is a CTA requirement to write IRFs in a common format with science tools.

The current format specification can be found here:  https://github.com/open-gamma-ray-astro/gamma-astro-data-formats

Currently *GammaPy* (http://gammapy.org) provides reader classes for these data, so perhaps contributing to that project such that they can be *written* would solve this problem.  Then we would just for example construct a `gammapy.irf.EffectiveAreaTable2D` object, and write it out in the correct format?

thomasarmstrong, 2017-11-15:
Hi Karl,
I have started adding some functionality to gammapy which should allow us to use it to write IRFs, I have just started with the current format that is found in the prod3b IRF files but I will go through the format specifications to make sure it is all compliant.

kosack, 2017-11-15:
@tino-michael this should then be used in your output, once Tom has finished.

TarekHC, 2017-11-15:
Hi @thomasarmstrong. I am interested in this, so if you need any help or for me to review the implementation, please let me know.

HealthyPear, 2020-05-20:
Hi all, I just bumped in this issue while roaming around....

We started developing [pyirf](https://github.com/cta-observatory/pyirf) and I will open some issues and projects there, to organize the work.
This issue will definitely be reproduced there.

It seems to me that given the situation it is also mutually exclusive, am I correct?
If this is right I think we can safely close this issue.

MaxNoe, 2020-05-20:
Yes, I think this no longer belongs here

closed 2020-05-20 by MaxNoe