cta-observatory/ctapipe #564: iminuit error during ctapipe installation

moderski, 2017-10-10:
Hi All,

I got an issue when trying to install ctapipe on mac with miniconda3 and python 3.6:
moderski$ conda install -c cta-observatory ctapipe
Fetching package metadata .............
Solving package specifications: .UnsatisfiableError: The following specifications were found to be in conflict:
  - ctapipe -> iminuit -> python 2.7* -> openssl 1.0.1*
  - python 3.6*
Use "conda info <package>" to see the dependencies for each package.

On my linux distribution I have iminuit 1.2 installed from the cta-observatory channel, but it seems that it is no longer there.
Is this a bug or there is a way around to install ctapipe?

Cheers,
Rafal

kosack, 2017-10-11:
THe problem is just that iminuit is not in the default anaconda channel.  You need to add the astropy or openastronomy channel to get it (perhaps we should add a copy of the package to the cta-observatory one, but then we'd have to maintain it).

```
conda config --add channels astropy
```

should fix that.  That's done automatically if you create your environment with the environment.yml file to install all packages.

closed 2017-10-11 by kosack

moderski, 2017-10-11:
Hi Karl,

Thanks. That worked.

RM