cta-observatory/ctapipe #567: Neighbour calibration

vuillaut, 2017-10-12:
Testing the neighbour calibration in ctapipe, here is what I obtain:
[image]

Some distribution shapes are unexpected (CHEC, DigiCam, NectarCam) and do not look physical. Errors might come from simulations also. 
Note that re-coding the neighbour calibration in C++ (HPC), we get similar results for LSTCam, CHEC and DigiCam but more realistic distributions for NectarCam and FlashCam.

For now I do not know where the issue might come from.

kosack, 2017-10-12:
Can you tell me what are these plots? Are tehy from a single pixel on a single camera, summed over events, or combined all pixels?  They look fine to me (mostly)  just recall the CHEC + Digicam use SiPms, where you see the 1,2,3,4,5 - up to 30PE peak if you have stats, whereas the regular PMs get washed out. Flashcam looks too wide, but that is to be expected since we don't yet support it's method, or perhaps there is a bug.

vuillaut, 2017-10-12:
Hi.
Sorry I should have said that these plots are the result of the neighbour method calibration for the whole cameras and for many events. The Y axis is normed.
I expected something like LSTCam.
- FlashCam is much wider that what we get with our algo
- The sharp shape at ~ -2pe for the NectarCam looks weird to me and we don't get it either with our calibration

Here is the comparison:
[image]

kosack, 2017-10-12:
I've never seen the sharp edge in nectar cam with any of the data I've looked at.  I just tried and can't reproduce it, though if the NSB is turned off, I would expect to see a sharp cutoff, since you cannot have less than 1 PE at the photocathode - you only get a smooth/negative distribution if there is some NSB on top being subtracted.  Which data files are you using? It definitely looks like a problem in one or the other code.  ASTRICam is of course normal, since you can't do any trace integration on it, so the NeighborPeak scheme is not usable. 

@watsonjj how much testing / cross-checks have you done with non-CHEC camera data with NeighborPeak integration?   Certainly there is something strange going on here in all cameras (none agree well).   One easy thing to do is to test it against what is done in `read_hess` or `

kosack, 2017-10-12:
see for example https://github.com/cta-observatory/ctapipe/blob/master/examples/notebooks/check_calib.ipynb

kosack, 2017-10-12:
Also, can you post a link to your neighbor-peak integration code? It's hard to compare without knowing if the algorithm is the same (there are many ways to do it)

watsonjj, 2017-10-13:
This is a little weird...I agree. It is like you are seeing the low illumination spectra of the camera...but you have masked out the cherenkov shower... so the only thing that should remain is the NSB, which shouldn't be coincident in neighbouring pixels, and so it shouldn't be visible with the NeighbourPeakIntegrator...

Using that notebook, it certainly as visible when using a LocalPeakIntegrator, making this even more stranger...

I personally haven't checked the integrators against read_hess. But Rachel had back when she worked on the code. Though, I do imagine they may not return the same values anymore, as I reinterpreted the code to achieve a similar method utilising numpy.

watsonjj, 2017-10-13:
Could I have the sim_telarray file you are using?

maxnoe, 2020-10-26:
@vuillaut @watsonjj This is a very old issue about a calibration artifact. Much has changed since then.

I will close this, feel free to reopen if you think this is still an issue. But at least a current comparison should then be included.

closed 2020-10-26 by maxnoe