cta-observatory/ctapipe #569: add author info to files

kosack, 2017-10-12:
For files where there is one or a few primary authors/maintainers, should we add an `__author__ = 'name'` tag at the top of the file? This may make it easier to track who to contact when making modifications.

e.g. from PEP8:

> Module level "dunders" (i.e. names with two leading and two trailing underscores) such as all , author , version , etc. should be placed after the module docstring but before any import statements except from future imports

So module code could look like:

*somemodule.py*

```python3
""" Implementation of some algorithm"""

__author__ = 'A. Author <author@place.org>'

from ctapipe.io.hessio import hessio_event.source
#...
```

Perhaps avoid adding email addresses, though, since that can increase spam since they are searchable online.

MaxNoe, 2017-10-13:
`git blame`

kosack, 2017-10-17:
git blame works, but it's not quite the same thing as maintaining a list somewhere of who is responsible for what. For example, somebody may come in and clean up a file as we did at the last workshop, and then many of the lines will "belong" to them, even though they did not write the code.  We do need some way to say "who should I ask about the implementation of this module?".  Perhaps it's not a critical problem yet.

watsonjj, 2017-10-17:
I would welcome this change. I think it should be a policy where if a PR is made on the code someone maintains, they are mentioned in the PR and are given an opportunity to comment on the changes. This would help to avoid changes that unintentionally break code.

MaxNoe, 2017-10-17:
@watsonjj Github automatically suggests the people who edited the code or the files before as reviewers.

kosack, 2017-10-27:
perhaps then having:

```py3
__maintainer__ = 'Some Person'
```
would be the best way then, and then just use GitHub/blame to get the full list.

MaxNoe, 2017-11-23:
This might also be interesting in this context: https://github.com/blog/2392-introducing-code-owners

watsonjj, 2017-11-23:
That sounds like the best way to achieve what I had hoped for (including the "Require review from code owners"). Thanks for that information!

maxnoe, 2023-01-04:
People come and leave, so I am not sure we should have these dedicated code owners. In the last 5 years there wasn't really a need I'd say, the number of core maintainers is small, that we basically review all PRs. I am closing this for now, might revisit in future.

closed 2023-01-04 by maxnoe