cta-observatory/ctapipe #571: Support event-wise and column-wise data with Containers?

kosack, 2017-10-18:
Right now the design of the `core.Container` class is a sort of *schema* defining what is in an "event", but it can also be considered to be a table description, where each data item is a column in an output table (see e.g. `ctapipe.io.HDF5TableWriter`, which converts event-wise data in `Container` classes to tables).   In this way it is very similar to *object-relational-mappers* (ORMs) like those used in database libraries. 

There are two issues:

 * different parts of the pipeline have different data-access use cases: some want to access a full event at a time (e.g. reconstruction), and some several *columns* of data for all events (e.g. event discrimination, IRF generation, and some parts of camera calibration involving pixel-wise distributions)
* our current MC data format supports only event-wise reading, so making columnar data access requires looping over events and building up a list. However, we can anticipate that the final CTA data format may allow for both modes of access. For example HDF5, or prototype formats like *pRun*, for even FITS tables, where data can be read row-wise (by event), or column-wise (a subset of data by time), or even by "hypercube", meaning some more complex slice through the data.

Right now, we use 2 ways of accessing data: from calibration to reconstruction, we use the container classes *event-wise* and write out HDF5 tables. Then in the rest of the pipeline, we access those tables via *pytables* interface, and ignore Container classes.  It looks something like this:

```py3
source = xxx_event_source(filename)
writer = XXXTableWriter(out_filename)

for event in source:
     do_something(event.dl1)  #event.dl1.tel[1].image is shape (npixels,)
     writer.write(event.dl1)
```   

later...

```py3
data = tables.open(out_filename)
images = data.DL1.tel_001.image  # data.DL1.tel_001.image is shape (npixels,nevents)

do_something_with_all_images(images)
```

So the question is: **is this ok, or should we try to support both methods of access in one API?** (anticipating again nicer data formats where we could, e.g. make a histogram of the pixel value for all events without doing an event loop).

I suspect this could be handled in a way where either you get a `Container` where each item inside represent an event-wise value.  For example *dl1.image = the array of integrated pixels for one event* with `shape=(npixels)` ), or you could request a data source the provides you the column-wise access, where `dl1.image` is *the images for all events* with `shape =  (npixels,nevents)`.

Any suggestions for a good design for this? Or should we just ignore it and only use Containers when we want event-wise, and another API when we want column-wise data (as is done now?)  @jacquemier  this relates of course to the data format issues you are looking into.

dneise, 2018-03-19:
I believe one way to think about this, is from an agile-development point of view:

KISS principle:
I would interpret KISS here as: do whatever solves your problem in the most simple way possible. 
So when thinking events use a row-wise interface. And when thinking stats, use a col-wise one. Done.

Another way to think about this, might be a performance point of view:

I have yet to find a (compressed) storage backend, which does not have a performance penalty on either row or columwise reading or writing. So I think data storage always specializes:

 * quick to write, but huge.
 * okay write speed and compressed but slow column-wise.
 * compressed and quick in both col and row access thanks to a sophisticated indexing system, but slow to write since indices need to be synced and not cross-language since only one closed source implementation in haskell exist...

And so on.

I believe correctly guessing what kind of access later will be performed most of the time on some output of mine, is often impossible. But keeping it easy, cross-language, and small (i.e. compressed) is often possible.

kosack, 2018-03-21:
So far what we have is: Containers represent single events, and once we go to the DL2 level of data processing, we abandon them and just read the data tables in their entirety and use `pytables` or `pandas` for the rest of the analysis.  This is ok with me, and maybe is the simplest thing, rather than trying to shoehorn everything to use Container classes. 

What I had though is that we could always make a function that loads all data columns of an output file (using pytables, etc) into a Container class, in which case the container would represent columns, not events. But perhaps that is just confusing.

maxnoe, 2021-05-05:
I think in recent times we basically decided that containers are for event wise quantities and we use astropy tables for groups of events.

So I am closing this. Feel free to reopen, if this should be discussed further.

closed 2021-05-05 by maxnoe