cta-observatory/ctapipe #577: Pixel-wise histogramming

calispac, 2017-11-14:
Hi all,

I am wondering if there is any plan to have a class/function to create an histogram along a given axis (e.g. pixel-wise).
#### Example on 2 waveforms: 
`data_cube = np.array([[1, 2, 2, 2, 5] ,[6, 7, 7, 8, 10]])
bins = np.arange(0, 11, 1)
hist = utils.histogram.1DHistrogram(bins=bins)
hist.fill(data)
hist_data = hist.data
print(hist_data)
`
[[0, 1, 3, 0, 0, 1, 0, 0, 0, 0, 0], [0, 0, 0, 0, 0, 0, 1, 2, 1, 0, 1]]

Right now the workaround I have for that in numpy is : 

`
hist_data = np.zeros((data.shape[0], bins.shape[0]))
for pixel in range(data.shape[0]):
     hist_data[pixel] +=  np.histogram(data[pixel], bins=bins)[0]
`
Of course this is taking much more time than if there was in numpy anything like : 
`
hist_data = np.histogram(data, bins=bins, axis=0)
`
I believe this could be useful if we want to do per pixel ADC distributions (for instance single p.e. per pixel or to check if a pixel is noisy). This might be linked to the point rose by @kosack in #571.

The issue with the actual event-wise reading and pixel-wise histogram construction is that it could have a huge performance impact if we need e.g. 10k waveforms of ~50 samples form ~ 2k pixels to build pixel-wise distributions.

Do you have any suggestions ? Or better workaround I missed ?
Thanks, Cyril

calispac, 2017-11-14:
This could do the job ?
```
data = np.random.uniform(size=(1296, 50))
bins = np.arange(0, 4096, 1)
temp = np.digitize(data, bins=bins)
temp = temp[..., np.newaxis] == np.arange(bins.shape[0])
hist_data = temp.sum(axis=1)
```

kosack, 2017-11-15:
One of the philosophical difference of numpy/scipy that is different from ROOT is that histograms are not considered 1st class objects, rather just operations using NDArrays like anything else.  So I think @calispac's suggestion is fine - we could of course add a nice function that wraps that for simplicity, since it may be something we do often.  

Even the `utils.Histogram` was just there as a stop-gap for ROOT users, however it's somewhat nice to include the I/O functions in it, so having a class to do what is shown above could still be useful, since we can add `read()` and `write()` methods to it if we like.

calispac, 2017-11-20:
The example above is not in-place and can take more computation time since complexity is O(n * n_bins) I ended up doing some benchmarks with various implementations: 

1. Python for loop over np.histogram
2. np.digitize + C code to replace np.bincount
3. full C code to compute the histograms

Finally the C code implementation performs better and complexity is O(n * log(n_bins)).

``
import numpy as np
import ctypes
from numpy.ctypeslib import ndpointer
import os
from matplotlib.offsetbox import AnchoredText
import matplotlib.pyplot as plt

lib = np.ctypeslib.load_library("histogram_c.so", os.path.dirname(__file__))
histogram = lib.histogram
histogram.argtypes = [ndpointer(ctypes.c_float, flags="C_CONTIGUOUS"),
                       ndpointer(ctypes.c_uint, flags="C_CONTIGUOUS"),
                       ndpointer(ctypes.c_uint, flags="C_CONTIGUOUS"),
                       ndpointer(ctypes.c_uint, flags="C_CONTIGUOUS"),
                       ndpointer(ctypes.c_float, flags="C_CONTIGUOUS"),
                       ctypes.c_uint, ctypes.c_uint, ctypes.c_uint]

__all__ = ['Histogram1D']

class Histogram1D:

    def __init__(self,  bin_edges, data_shape=(1, ), name="1D Histogram", axis_name=None):

        self.data = np.zeros(data_shape + (bin_edges.shape[0] - 1, ), dtype=np.uint32)
        self.shape = self.data.shape
        self.bins = np.sort(bin_edges).astype(np.float32)
        self.bin_centers = np.diff(self.bins) / 2. + self.bins[:-1]
        self.n_bins = self.bin_centers.shape[-1]
        self.name = name
        self.axis_name = axis_name
        self.underflow = np.zeros(data_shape, dtype=np.uint32)
        self.overflow = np.zeros(data_shape, dtype=np.uint32)

    def fill(self, data_points):

        if data_points.shape[:-1] != self.shape[:-1]:

            raise IndexError

        new_first_axis = 1
        for i in self.shape[:-1]:
            new_first_axis *= i

        data_points = data_points.reshape(new_first_axis, -1).astype(np.float32, order='C')

        self.data = self.data.reshape(new_first_axis, -1)
        self.underflow = self.underflow.reshape(new_first_axis, -1)
        self.overflow = self.overflow.reshape(new_first_axis, -1)

        histogram(data_points, self.data, self.underflow, self.overflow, self.bins, data_points.shape[0], data_points.shape[-1], self.bins.shape[-1])

        self.data = self.data.reshape(self.shape)
        self.underflow = self.underflow.reshape(self.shape[:-1])
        self.overflow = self.overflow.reshape(self.shape[:-1])

    def errors(self, index=[...]):

        return np.sqrt(self.data[index])

    def mean(self, index=[...]):

        return np.sum(self.data[index] * self.bin_centers, axis=-1) / np.sum(self.data[index], axis=-1)
[... 112 lines cut ...]
                    index_bin = (first + last) / 2;
                }

            }

            else if(data[index_data] < bins[0])

            {

            ++underflow[i];

            }

            else if(data[index_data] >= bins[n_bins - 1])

            {

            ++overflow[i];

            }

        }
    }
}
``

kosack, 2017-11-21:
see also https://github.com/astrofrog/fast-histogram

kosack, 2017-11-22:
I jsut want to avoid adding too much code that we have to maintain just to get a small optimization. Right now ctapipe is intended for a design study, not for being ultra fast (optimization will come later).  

I get resonable speeds using np.histogram as follows:

```
n_pix, n_images = 2048, 10000
images = np.random.normal(shape=(n_pix, n_images))  # a bunch of random camera images

# now make histograms for each pixel, over all images:

hists = np.apply_along_axis(np.histogram, axis=1, arr=images, range=(-5,5), bins=50)
```

On my machine, this takes *180 ms* to make 2048 histograms over 10000 images. It could clearly be faster, but do we really care at this point? I'd prefer simpler/smaller code.

kosack, 2017-11-22:
This may imply that the `ctapipe.utils.Histogram` class is not always the right tool for the job - just using simpler np.histogram (or `np.histogram2d, np.histogramdd`) could be better in cases where there need to be a lot of histograms, and each doesn't need to be stored as a separate FITS file (which is what `ctapipe.utils.Histogram` was designed for)

calispac, 2017-11-22:
I agree that simpler is better. A use case would be to have (n_pixels, n_images) ~= (2000, 50). Fill the histogram then proceed to the next event (of same shape). 
The proposition above has good performances if n_image is large. For realistic values of n_images =~ 50 it would require to make batches of events before filing the histogram. 
I don't think that the user should handle such batches to optimize performances.

Actually for DigiCam calibration we used to do exactly what you proposed and ended up doing some batches in the event loop.

If there is a way to access the data row-wise then the proposition above is fine I guess but you still need a lot of memory.

kosack, 2017-11-22:
yes, this may be resolved in fact by the choice of data format - some of the proposed DL0 formats allow efficient column-wise access (all images for all events, for example), in which case it's fast to do histogramming this way.  If we do it always event-wise, then we should come up with a faster solution.

TjarkMiener, 2026-04-20:
We are working towards the support of `SPE fitting` within `calibpipe`, therefore we need to aggregate pixel-wise histograms. The idea would be to add a `HistogramsAggregator(BaseAggregator)`. Nectarchain is computing historgrams using `numba` (with just a few lline of codes) but I'm not so sure if `numba` really speed up things when calling `np.histogram()` (see [here](https://github.com/cta-observatory/nectarchain/blob/main/src/nectarchain/makers/component/charges_component.py#L58-L114)). Though not sure if we need tons of custom code to try to speed up this calculation which might be not worth in the end. I will start migrating the nectarchain function and do some performance testing.