cta-observatory/ctapipe #579: Installation description for users

justuszorn, 2017-11-17:
If following the installation descriptions for users on the [main page](https://github.com/cta-observatory/ctapipe) ctapipe is not installed properly, so import ctapipe in python does not work.

Instead of 'conda install ctapipe' as las step in the installation procedure, it should be 'conda install -c cta-observatory ctapipe' as described [here](http://cta-observatory.github.io/ctapipe/getting_started/).

MaxNoe, 2017-11-17:
I think the channel should be `conda-forge` since a few releases, so `conda install -c conda-forge ctapipe`

justuszorn, 2017-11-17:
Well it worked with `conda install -c cta-observatory ctapipe`

kosack, 2017-11-21:
No, it should be cta-observatory not conda-forge (I have not uploaded ctapipe to conda-forge for building).  

However, if you follow the main page instructions and use the `environment.yml` file, it should work: you should have the cta-observatory channel added automatically by the yml file, and so `conda install ctapipe` should work... Can you tell me what is happening? Are you sure you are in the virtual environment that was created ("cta" in the example?). 

Perhaps some versions of conda do not add the channels listed in the yml by default... in which case  we could add "conda config --add channels cta-observatory"

kosack, 2017-11-21:
just to note: conda-forge is also needed, but only for the other non-cta dependencies like the latest astropy and iminuit packages

kosack, 2017-11-21:
In any case, I added the channel explicitly now on the instructions, so it should work regardless if the channels in the YML files get added to the list of defaults.

closed 2017-11-21 by kosack

justuszorn, 2017-12-13:
Just a minor comment: If I follow the instructions for developer and do `git clone https://github.com/[YOUR-GITHUB-USERNAME]/ctapipe.git`, I get a file called _cta-env.yml_ and not _environment.yml_, thus `conda env create -n cta-dev -f environment.yml` won't work...

dneise, 2017-12-13:
The current master contains a file called `environment.yml` 
you can check that here  https://github.com/cta-observatory/ctapipe

When I look at [your fork](https://github.com/justuszorn/ctapipe), I see indeed a file called: `cta-env.yml`.

The fork says: "This branch is 2388 commits behind cta-observatory:master. "

Maybe it would be a good idea to sync your fork with the upstream master again.
https://help.github.com/articles/syncing-a-fork/

justuszorn, 2017-12-13:
Oh yes, sorry, thanks for your help!