cta-observatory/ctapipe #581: docu: Missing Link to "CTA Code Standard" document

dneise, 2017-11-23:
I am new to CTA and just starting to follow the nice ctapipe documentation.

I believe CTA documentation does partly exist outside the cta sphinx docu, for newcomers, having explicit links to the relevant documents would help a great deal.

https://github.com/cta-observatory/ctapipe/blame/master/docs/development/code-guidelines.rst#L4

kosack, 2017-11-23:
All official docs are in the CTA Jama system, but I don't think the code-standards one is done being reviewed by the PC, so so far we don't have anything to link to.  Hopefully that will come soon! It's very hard to find documents in CTA (which can be in redmine, sharepoint, Jama, or elsewhere). We hadn't really considered the ctapipe docs to be the central source, but it's a good idea to link things when possible (even if some are internal-only)

dneise, 2017-11-23:
>  It's very hard to find documents in CTA 
I agree entirely. Which I think makes it even more important to clearly link to documents one refers to, and not just mention their "name".

> We hadn't really considered the ctapipe docs to be the central source, but it's a good idea to link things when possible (even if some are internal-only)

I understand. What would you consider a valid starting point for newcomers, who intend to collaborate on the telescope/camera specific software?

kosack, 2017-11-23:
If it's software that will go into the off-line pipeline (e.g. part of ctapipe), then start with the python style guide in the ctapipe docs (in fact the official guide is based on this one so far, so it should be essentially the same).  For other camera software (C/C++ or Java), refer to the ACS style guides for now, I think, since that is where that software goes.   I'll check on the status of the official doc.

closed 2017-11-23 by kosack