cta-observatory/ctapipe #582: docu: Missing Link to "CTA Data Level Models definition document"

dneise, 2017-11-23:
For the same reasons, I meantioned in #581, I think here a direct link to the document in question could possibly help the reader a lot.

https://github.com/cta-observatory/ctapipe/blame/master/docs/calib/index_camera.rst#L27

kosack, 2017-11-23:
Can we link to documents inside Jama? I guess there is probably a way, but of course they are not public.

watsonjj, 2017-11-23:
Yes good suggestion.
I believe this is the relavant link that should be added:
https://jama.cta-observatory.org/perspective.req#/items/26528?projectId=6

MaxNoe, 2017-11-23:
I'm confused. I cannot login in there with the account that works for the indico. What account exactly do I need to use and how do I know?

dneise, 2017-11-23:
I can login here and see the document:
https://jama.cta-observatory.org/perspective.req#/items/26528?projectId=6

dneise, 2017-11-23:
>  but of course they are not public.

I think this is an issue, how can one expect collaborators to adhere to rules, which are regarded as top secret. But it is an issue, which cannot be solved by ctapipe. 

I believe all one can hope for, is that anybody reading the ctapipe docu, has somehow also been awarded with access to the sacred archives of CTA.

AMWMitchell, 2017-11-23:
@MaxNoe Logging in to Jama works with the usual CTA login. However, users have to be added manually (!) from the LDAP database, so we tend to add people on request. I just added you so you should be able to login now.

kosack, 2017-11-23:
> I think this is an issue, how can one expect collaborators to adhere to rules, which are regarded as top secret. But it is an issue, which cannot be solved by ctapipe.

It's mainly because the project management didn't think about software until recently, so we're a bit late on having official software docs. There are people working on them though. Generally all CTA info is internal, ctapipe and related packages is only public since we didn't have a github or gitlab instance to use internally, and there is no real reason to make it private. However, internal CTA discussions should be kept on the internal pages, of course.

kosack, 2017-11-23:
I added the link to the document

closed 2017-11-23 by kosack