cta-observatory/ctapipe #587: Use of custom Exceptions?

dneise, 2017-11-27:
I stumbled over [pixel_likelihood.py](https://github.com/cta-observatory/ctapipe/blob/master/ctapipe/image/pixel_likelihood.py)

And saw custom Exceptions are used in places where the built ins seem to be just fine:

E.g. here https://github.com/cta-observatory/ctapipe/blob/master/ctapipe/image/pixel_likelihood.py#L130

PixelLikelihoodError is raised, which derives from probably the most general of all specific Exceptions: (RuntimeError)[https://docs.python.org/3.5/library/exceptions.html#RuntimeError]

One could also have raised (ValueError)[https://docs.python.org/3.5/library/exceptions.html#ValueError] in this context.

----

In the next case one could think about emitting a warning, not raising an exception `if np.any(np.isnan(first_term))`. 
The result might be fine for 99% of all pixels and only be np.nan for the single pixel, which is broken and therefor nan is even expected. 

https://github.com/cta-observatory/ctapipe/blob/master/ctapipe/image/pixel_likelihood.py#L154

-----

The 3rd case looks again like a ValueError to me:

https://github.com/cta-observatory/ctapipe/blob/master/ctapipe/image/pixel_likelihood.py#L345

dneise, 2017-11-27:
There is sometimes a trend visible to an abundance of custom exceptions ... I was wondering what the policy regarding this is in ctapipe.

c.f. https://softwareengineering.stackexchange.com/questions/275631/when-should-i-subclass-an-exception-in-python

dneise, 2017-11-27:
Another example: In this case:

https://github.com/cta-observatory/ctapipe/blob/master/ctapipe/utils/CutFlow.py#L111

The custom Exception "UndefinedCut" is raised instead of a built in KeyError.

kosack, 2017-11-27:
I sort of like having well-defined exception names for cases where the exception may be ambigious and handled different, so they can be traced easier (the message is not always sufficient since it is only human-readable), but I agree they should at least be a subclass of the most applicable built-in (like `ValueError`, if it is an error in value).

dneise, 2018-03-19:
Thank you for the clarification. 

I think this matter is solved.

closed 2018-03-19 by dneise