cta-observatory/ctapipe #588: Does Trigger container should include camera trigger information ?

calispac, 2017-11-28:
I am wondering if the `DataContainer` should include trigger camera information such as 

- Trigger threshold (ADC count)
- Trigger algorithm (int flag)

In this case should we have a `Field` in DataContainer with

`trig = Field(TriggerContainer())`

with `TriggerContainer` beeing

`class TriggerContainer(Container)`
`central_trigger = Field(CentralTriggerContainer())`
`local_trigger = Field(Map(LocalTriggerContainer()))`

with `Local` beeing related to camera/telescope level and `Central` at the array level.

The `LocalTriggerContainer` could also include some specific camera trigger informations such as:

- time above threshold
- bool array of the part of the camera that where above threshold
- trigger input/ouput traces (DigiCam specific although I don't know if other cameras provide this information)
- pixel clipping level
- patch clipping level
- disabled parts from trigger algorithm

kosack, 2018-01-04:
Yes and no - we're currently removing everything from `DataContainer` that doesn't change *per event*, and everything else goes into a `ctapipe.instrument.InstrumentDescription` class (like `OpticsDescription`.   Basically: *Containers are only for things that need to be read/written  on an event-wise basis* (they act as schema for an output table).  Data realted to instrumental configuration gets specialized classes. 

however, we definitely *do* need an `instrument.TriggerDescription` class.   Right now that information cannot be read from the MC files, so it wasn't high on the priority list.  If you have a suggestion for a structure for this, that could help - I'm not sure what information is needed.

kosack, 2018-01-04:
I just noticed that some of the data you show above, do change per event.  In that case there should be some trigger container as well. However, we may fold some of that into a packed pixel_info field in the  DL0Container, since we cannot store all of that information in the DL0 camera files without blowing up the data rate.   I'd wait until we have the DL0 format spec before implementing anything. 

In the current draft of the DL0 format description (which I'm working on now), you get 6 bits per pixel per event to store any pixel-wise trigger info. Is that sufficient?   The main problem that might occur is if we have info that is not pixel-wise, but patch-wise, and the patches overlap. In that case we may need to use some more bits, but currently the file sizes are close to being above the physical limits of the off-site link.

calispac, 2018-01-08:
Here I see two scenarios.
1. The trigger is fixed for a night/run.
2. The trigger is adjusted during data taking.

In case 1. then the information that could be stored at DL0 (event-wise) are:

- trigger algorithm type (muon trigger, trigger 1, trigger 2, etc.)
- time above threshold
- patches above threshold (non overlapping)
- (the 2 points above could be inferred from a 1-bit waveform per patch, showing the "trigger" waveform, i.e. trigger output traces)

Concerning trigger input traces (trigger waveform which is directly compared to the threshold), I suppose it is not important for reconstruction. It provides info for the "experts". So this is not supposed to go to DL0 but rather R1. Beside it is very camera specific.
The rest could be stored in the `instrument.TriggerDescription`.

In case 2. everything is potentially changing event-wise. Since the file size is limited we can have a field in DL0 referring to `instrument.TriggerDescription` for the correct trigger settings associated with the DL0 `Container`.

Here is an estimate of the amount of trigger data required at DL0 (for DigiCam).

- trigger algorithm type (8 bits)
- time above threshold (n_samples x 1 bit = 50 bits)
- patches above threshold (n_patches x 1 bit = 432 bits)
- trigger output traces (n_samples x n_patches x 1 bit = 21600 bits)

If there is 6 bits per pixel available in DL0 then it makes 6 x 1296 pixels = 7776 bits for the camera. Therefore trigger output traces can not be in DL0. For me it makes also sense to put it in `R1` as for trigger input traces. Also this information can be recomputed latter having a description of the trigger algorithm. One just needs the patches mapping, the baseline used in this event and the pixel waveforms.

kosack, 2018-01-09:
@calispac thanks, that's useful to know. Yes, I think then in R1 we can put anything you like (as long as it doesn't saturate the camera-to-actl fibres), and then in DL0 it can only be the reduced info (no trigger traces, etc). Anything that doesn't change per event, can just go into `instrument`.  I think in DL0 all we can store is what patches triggered for a one or two thresholds.  The main question is whether or not that info is really useful per event, or whether it could become DL0.TEL.MON data, where it is stored *stastically* at 1Hz or so. E.g. the main question is: do you really need to know the exact trigger pattern per event, or could you get away with knowing that a trigger patch only e.g. triggers 1% of the time compared to neighbor patches (to identify problems)?

calispac, 2018-01-10:
I am not an expert on array trigger but I would assume that the array trigger needs to weight each trigger it receives from the camera. Doing so the array trigger can target specific telescopes to send a trigger.
Calculation of this weight could depend on : 

- time above threshold
- spread of the trigger among camera

Concerning monitoring data, I agree that to identify non uniformity in the trigger you only need to store for instance the number of triggers per patch durring ~ 1 second.

maxnoe, 2020-10-26:
The CameraTriggerContainer now contains the trigger pixels.

closed 2021-05-25 by maxnoe