cta-observatory/ctapipe #591: should `Component` implement `Processor` interface

dneise, 2017-11-30:
Looking at the [picture about the core structures](https://cta-observatory.github.io/ctapipe/core/index.html) one might get the impression, that `Component` implements some kind of "Processor" interface. 
By "Processor Interface" I mean some specific way to implement the incoming and outgoing arrows in the image. 

To give some examples of frameworks, I happen to know about, which also utilize this concept, let me refer to the `streams.Processor` in the [FACT analysis framework: fact-tools](https://github.com/fact-project/fact-tools). A simple example implementation could be the (MaxAmplitude Processor)[https://github.com/fact-project/fact-tools/blob/master/src/main/java/fact/extraction/MaxAmplitude.java#L20] which provides a `process` method, which receives a very general "input" of type "Data", which is really nothing else than a Python `dict` representing the smallest independent unit for the analysis, which is in case of FACT a single event. Within the process method, the processor extracts whatever it needs from the `input` ... performs a calculation and appends the result into the dict and returns it. This implementation certainly is not perfect, but it shows what I mean by processor interface. By virtue of implementing a common interface, one can describe the "pipeline" or "flow" using e.g. a markup language. fact-tools uses e.g. XML.

Another framework, which also implements this idea is the [FACT analysis framework: MARS](https://trac.fact-project.org/browser/trunk/Mars). The Base class in MARS, which defines the interface is to my knowledge (MTask)[https://trac.fact-project.org/browser/trunk/Mars/mbase/MTask.cc#L27]. Each Task, which derives from MTaks needs to implement the "Process()" method (and some more) to be useful inside the framework. Within the MARS framework, the input and output of Tasks is more structured than in the former example. Tasks read from certain Containers and write into certain Containers. This structure seems to be also important in ctapipe.

Now after this long pre-ample. I was wondering if you would also like to formalize the interface of a "Component" by letting it implement some kind of "processor interface". I have seen in the current master Component is an (abstract base class)[https://docs.python.org/3.6/library/abc.html] without actually having abstract methods (yet?).

I was looking at several implementations of Components which already exist. I found for example these in `ctapipe.calib.camera` which look a lot like processors to me. 

```python
    class CameraDL0Reducer(Component):
        def __init__(self, config, tool, **kwargs):
            # configures itself
        def check_r1_exists(self, event, telid):
            # checks that input exists
        def reduce(self, event):
            # performs its duty

    class CameraR1Calibrator(Component):
        def __init__(self, config, tool, **kwargs):
            # configures itself
        def check_r0_exists(self, event, telid):
            # check that input exists
        def calibrate(self, event):
            # perform the duty

    class CameraDL1Calibrator(Component):
        def __init__(self, config, tool, **kwargs):
            # configures itself
        def check_dl0_exists(self, event, telid):
            # check that input exists
        def calibrate(self, event):
            # perform the duty
```
Base on their structure I was wondering if you would be open to the proposal of for a (simplistic) Processor interface. 

The most simple is of course:
```python
    class Processor(ABC):
        @abstractmethod
        def process(self, event):
            # calculate something and **attach** it to event
            return event
```
And then, have `Component` inherit from it like this, e.g.:
```python
    class ConfigurableProcessorMeta(type(Configurable), type(Processor)): pass

    class Component(Configurable, Processor, metaclass=ConfigurableProcessorMeta):
        # ...
```

dneise, 2017-11-30:
Of course one can think about more interesting features of the `Processor` interface. We already brain-stormed about this a bit and thought these might be the features a `Component` should have to be most useful for us. 

 * Configurable (already done)
 * CanLog (already done)
 * HasProcessMethod (most simple proposal fulfils this)
 * ChecksForInput
 * DoesNotClobber (<-- is this wanted at all?)

The latter two could also be handled by a possible `Processor-ABC`. 
A Processor needs to know at design-time which part of event it is reading from and which part of event it is writing to. Then the base class could ensure the input exists and make sure output is never clobbered.

MaxNoe, 2017-11-30:
See https://github.com/cta-observatory/ctapipe/pull/236

dneise, 2017-11-30:
Okay, so this can be closed following the same arguments. Thanks.

MaxNoe, 2017-11-30:
The PR was a Year ago. A lot has changed since then. I still like the idea.

All processors should get an appropriate container as input and have a defined output. This way, one could even build dependency graphs between the processors and automatically determine order and parallelizability.

dneise, 2017-12-04:
Ah, I see. I assumed your comments goal was to show me, this issue was a duplicate and that I should close it.

kosack, 2017-12-07:
Just to note, the concept you describe  (processor chains) is actually implemented in the experimental `ctapipe.flow` module.

For now, the Component class's only purpose is to handle configuration data, not impose a data flow scheme, but I'm not against it - in fact that was the original design, but since we have still been discussion how to handle refactoring the event Container data structure, it seemed too early to base all flow on it.    The main reason was to avoid over-design too early, which caused so many bad and hard to fix design choices e.g. in the HESS software (which is completely "processor" flow based, as you describe). 

I always had the idea of multiple levels of abstraction:

* most code is just a single function (like `hillas_parameters()` or `tailcuts_clean()`), unless it needs some special state stored between calls, in which case it is simply a `Component` subclass, with functions in it

* Any higher-level flow-based data could wrap that concept, as in the `flow` module, or we could use fancier systems in the future (spark, or dask, whatever), but algorithm designers shouldn't have to worry about it.

Of course, in the end, it seems we are converging on a bunch of components that  work in a sort of similar fashion, so coming back to a simple processor model may be ok.  Does it actually solve any problems, though? Other than making the function call always called "process(event)"?

kosack, 2017-12-07:
Oh, I should say, I also originally wanted to avoid passing the "event" as a high-level data structure to every Component blindly.  That makes it so hard to debug where things in the data structure are updated, especially when the chain of processing becomes complex. That was one of the main criticisms of existing frameworks. But as usual, we seem to be doing that everywhere...

MaxNoe, 2017-12-07:
>  Does it actually solve any problems, though? Other than making the function call always called "process(event)"?

Yes, it makes it a lot easier to define analysis steps that should be applied, somthing like this:

```
Pipeline(
    source=HessioEventSouce('/path/to/file'),
    dl2steps=[CalcHillasParameters, CalcStatistics, CalcMyFancyFeature],
    sink=FitsWriter('path/to/file'),
)
```
So people do not need to implement a flow logic every single time they change the analysis.

> Oh, I should say, I also originally wanted to avoid passing the "event" as a high-level data structure to every Component blindly. That makes it so hard to debug where things in the data structure are updated, especially when the chain of processing becomes complex.

I don't think we can get rid of passing an array event around as a whole, but one can solve  most of the debug problems by having defined input and outputs (probably as containers) and unit tests.

dneise, 2017-12-07:
I must say, I really like your way of thinking @kosack 

> Does it actually solve any problems, though? Other than making the function call always called "process(event)"

You might be right, it might not solve any problems. I would argue, that @MaxNoe  proposed to make Processors callable so, the call would be: `cleaner(event)` not so much `cleaner.process(event)` which would is already less noisy.

However, I agree, even clearer would be something like
    
    image = integrate(adc_samples)
    shower = clean(image)

Since it can be clearly distinguished from:

    image = integrate(adc_samples)
    arrival_times = rising_edge(adc_samples)
    shower = clean(image, arrival_times)

The latter form clearly shows the dependencies of cleaner, and that it uses (probably) time and charge to make a decision, it can still ignore one of the inputs ... 

The "flow" version would hide that somehow, right?

    event = integrate(event)
    event = rising_edge(event)
    event = clean(event)
 
I think the flow version has more structure and allows thus for things like this. 

```python
#contents of: pipeline.py
in = EventIo
pipe = [Integrate, RisingEdge, Clean, Hillas]
out = Csv(Hillas)
```

And then a tool could look like this:

```python
#contents of: some_tool.py
import pipeline
# configure the processors using traitlets config 
processors = [proc(config) for proc in pipeline.pipeline]

with pipeline.out(config) as out:
    for event in pipeline.in(config):
        for proc in processors:
            # an interpretation of provenance
            logging.info("%r is now working on %r", proc, event) 
            event = proc(event)
        out(event)
```

This way, it is possible this way to have "processor developers" who only develop a certain processor, but in order to test/use it in the pipeline, they do not need to implement a tool, they just need to *configure* a tool. 

Of course you all know this, but I still wanted to write it down, so we see it with all its pros and cons. 
Repeating your argument from above, that `process(event)` does not tell the reader much, I would like to add that: `cleaner = Cleaner(config)` also does not tell me much about what parts of the config file are actually used to initialize that processor, I have to look into the implementation of Processor to learn that, while:
```
c = config['cleaning']
cleaner = Cleaner(c['threshold_core'], c['threshold_fringe'])
```

Clearly tells me, that this cleaner does not look at "threshold_arrival_time", which I just defined in the config file.

------

To sum this up: I think there are arguments for both styles. 

I like the idea to implement an analysis step as a simple function and then wrap the analysis into a Processor which is only unpacking the `event` into what the function needs and is putting the result of the function nicely into the correct container. This seems to allow developers to start in a simple way and later insert their implementation painlessly into the framework. 

It is not like this old story .. ah somebody wrote something and now we have to write it again "correctly" so it can run in our framework. Wrapping can solve this, quite nicely I assume.

MaxNoe, 2017-12-07:
Yes I agree. `Processors` should be thin wrappers around the underlying `work` function. LIke getting the correct input and encasing the output.

mackaiver, 2018-02-08:
Just to add another existing API.
I implemented it a while ago in this [project here](https://github.com/mackaiver/jayct). The basic interface looks like this:

        ImageReader events = ImageReader.fromURL(url);

        for (ImageReader.Event event : events) {
            List<ShowerImage> showerImages = TailCut.onImagesInEvent(event);
            List<Moments> moments = HillasParametrization.fromShowerImages(showerImages);

            ReconstrucedEvent reconstrucedEvent = DirectionReconstruction.fromMoments(moments, event.mc.alt, event.mc.az);

            if (reconstrucedEvent.direction.isNaN()){
                continue;
            }
            
            // continue doing whatever you do
        }

maxnoe, 2021-05-25:
We implemented `__call__` no for most components that work on events

closed 2021-05-25 by maxnoe