cta-observatory/ctapipe #592: MC non triggered events

thomasarmstrong, 2017-11-30:
I was trying to see if I could use ctapipe to read in sim_telarray data and get both the triggered data and all the simulated MC data. This would be used to calculate, for example, the trigger efficiency with respect to photoelectrons in the image. As far as I can see this isn't possible at the moment? 

Looking in hessio_event_source(...) in ctapipe/io/hessio.py

```python
with open_hessio(url) as pyhessio:
    # the container is initialized once, and data is replaced within
    # it after each yield
    Provenance().add_input_file(url, role='dl0.sub.evt')
    counter = 0
    eventstream = pyhessio.move_to_next_event()
    ...
```
I see that there could be an option to somehow switch to `eventstream = pyhessio.move_to_next_mc_event()` which would then provide you with all the simulated data. When testing this however, I came across something I don't quite understand. If I run some example code such as:

```python
source = hessio_event_source(file_path, max_events=None)
for event in source:
    print(event.r0.event_id, event.mc.energy, sum(event.mc.tel[1].photo_electron_image))
```
With the original code [move_to_next_event()]
```
1 0.7434797286987305 TeV 141
2 0.5755431056022644 TeV 223
3 0.32009783387184143 TeV 81
4 0.7020784020423889 TeV 218
5 0.2921573519706726 TeV 89
6 2.6281118392944336 TeV 1149
7 4.678738117218018 TeV 2734
8 0.3247097432613373 TeV 123
9 0.937703549861908 TeV 139
10 4.912286281585693 TeV 183
11 1.148128628730774 TeV 124
12 1.5528963804244995 TeV 359
13 0.3332664668560028 TeV 189
14 10.62702465057373 TeV 300
15 0.4885904788970947 TeV 227
16 14.384729385375977 TeV 925
17 0.42923223972320557 TeV 110
18 1.918776273727417 TeV 202
19 3.072026252746582 TeV 202
20 1.6568242311477661 TeV 172
```
and with edited code [move_to_next_mc_event()]
```
100 0.21774375438690186 TeV 0
200 0.326642781496048 TeV 0
300 0.7434797286987305 TeV 0
400 0.46060770750045776 TeV 141
500 0.4362963140010834 TeV 31
600 0.28762349486351013 TeV 31
700 0.5887979865074158 TeV 31
800 0.5755431056022644 TeV 28
900 0.4464867115020752 TeV 223
1000 0.42257633805274963 TeV 223
1100 0.42465123534202576 TeV 223
1200 0.2719481289386749 TeV 223
1300 0.9962665438652039 TeV 223
1400 2.7529940605163574 TeV 45
1500 0.23600557446479797 TeV 22
1600 0.39889732003211975 TeV 22
1700 0.4632396399974823 TeV 22
1800 1.089866042137146 TeV 63
1900 0.2184465080499649 TeV 15
2000 0.2836531102657318 TeV 15
```

If you take the event which has a simulated energy of `0.7434797286987305 TeV`, this has event number 4 and a total of 218 p.e. in the image in the first case, and event number 300 and a total of 0 p.e. in the second case (where the correct p.e. value seems to be associated with the next event).

Additionally if I try to calculate the trigger efficiency, I can only obtain a sensible answer if I remove all the duplicated p.e. values. 

This could be a non issue if there is a different way to read in the simulated data?

Sojourner92, 2017-12-01:
Hi Tom,
FACT have this bit of code that might help called pyeventio (https://github.com/fact-project/pyeventio), you could theoretically read out the MC energies from the corsika.gz file you ran simtelarray with. Perhaps it'd be worth integrating this into ctapipe?

kosack, 2017-12-05:
We hadn't considered supporting non-triggered MC events, so there is probably a logic error when simply changing to the `move_to_next_mc_event` method when filling the containers. It's never been tested.   My guess is it's something simple that we could fix however.  If you would like to commit this and debug it, please open a pull-request.

You can also in principle get a histogram of the simulated events from the MC file, but we don't yet have an accessor method in `pyhessio` to get histograms (there is an open issue about that: cta-observatory/pyhessio#49 ).

MaxNoe, 2019-01-03:
pyeventio now fully supports simtel files, you can should be able to get every information you need.

We also moved pyeventio to `cta-observatory` and you can `pip install eventio`.

To get information about all simulated events, loop through a file and look at all `MCShower` and `MCEvent`. 

There is one `MCShower` per simulated shower and then for each shower there are number of reuse `MCEvent` which basically just contain the impact position.

See this example: https://github.com/cta-observatory/pyeventio/blob/master/examples/iter_mc_events.py

closed 2019-01-03 by MaxNoe