cta-observatory/ctapipe #593: Code smell and Possibility to reduce coupling between `Components`?

dneise, 2017-12-04:
I reviewed `core.Component` again and got the impression that the `parent` member might have some issues: 

1. Smell: `parent` reminds one of inheritance, but this is a different relationship.
2. Coupling:  Two otherwise possibly unrelated things, get coupled here, just to share logging setup.
3. Surprise: `logging` (loggers are arranged in a namespace hierarchy)[https://docs.python.org/3/howto/logging.html#advanced-logging-tutorial], effectively sharing the configuration of their ancestors in this hierarchy. 

As Ward Cunningham put it: 
> You know you are working on clean code when each routine you read turns out to be pretty much what you expected.

You see how old this quote is, by Wards use of the term "routine", but I think it still applies. People used to the `logging` module, know there is no need to propagate loggers down form the application into the depth, since all child loggers inherit their settings from the ROOT logger. So for them giving a `parent` to each Component is quite surprising I'd assume. 

Given the general name `parent`, one might assume a specific Component instance and its `parent` might share more than only the logging configuration. I tried to search for it in the entire code base, but found nothing, so this "parent-child-relationship" is really only about logging. 

Also the documentation of [Component](https://cta-observatory.github.io/ctapipe/api/ctapipe.core.Component.html#component) clearly states that there is a parameter called `parent` but it does not explain that parameter. 

The doc-string of `__init__` explains parent as: "Tool or component that is the Parent of this one"

----

So I was wondering if this `parent` parameter could possibly be dropped entirely. This would solve all issues in one strike.

MaxNoe, 2017-12-04:
What components do is described in their docstring: https://cta-observatory.github.io/ctapipe/api/ctapipe.core.Component.html#ctapipe.core.Component

The central idea is, that they are configurable objects, and the configuration is made via a central config and command-line arguments of the calling `Tool`. 
The idea is, that the configuration makes up a single tree, which is important for provenance and reproducibility.

> Smell: parent reminds one of inheritance, but this is a different relationship.
It's a configuration `tree`, the node above is called parent. I think this is the correct word.

> Coupling: Two otherwise possibly unrelated things, get coupled here, just to share logging setup.
This is not true. It's as much about the global configuration as about logging.

> Surprise: logging loggers are arranged in a namespace hierarchy, effectively sharing the configuration of their ancestors in this hierarchy.

I don't get it. The `Component` makes sure, that the logging hierarchy reflects the Component hierarchy. That's a good thing imho.
https://github.com/cta-observatory/ctapipe/blob/2bff09fac5a0119a4b09ea1fd22bd484232581dd/ctapipe/core/component.py#L72-L73

dneise, 2017-12-04:
I talked with @MaxNoe on the phone about this, and he me explained a lot.

I still have a question about this, and will simply close this issue and open an new one.

closed 2017-12-04 by dneise

dneise, 2017-12-04:
> Given the general name parent, one might assume a specific Component instance and its parent might share more than only the logging configuration. I tried to search for it in the entire code base, but found nothing, so this "parent-child-relationship" is really only about logging.

Clearly this was my mistake: I overlooked this line:
https://github.com/cta-observatory/ctapipe/blob/2bff09fac5a0119a4b09ea1fd22bd484232581dd/ctapipe/core/component.py#L68

kosack, 2017-12-04:
Essentially, `Component` and `Tool`  are just subclasses of `traitlets.config.Configurable` and `traitlests.config.Application` with some nicer defaults.  Those base classes also have some confusing design decisions, but were the easiest things to start with without re-writing something custom, and seem to work fine for ipython, jupyter, and other projects that use them.  

For the parent relationship, so far it was just for logging and config, but perhaps it's not necessary to keep such a tree. The parent is indeed passed to the `Configurable` superclass, but I'm not sure if it's fully necessary (the docs of `traitlets` are less than desirable).  If you can suggest a better way, we should certainly simplify it!