cta-observatory/ctapipe #594: What is the purpose of the `name` class.variable?

dneise, 2017-12-04:
I see in some classes something like this
```python
class XXX:
    name = 'XXX'
    # ...put implementation here ... 
```

So in this case `XXX.name == XXX.__class__.__name__`, but there must be some semantic difference between `self.__class__.__name__` and this class variable `name`.

Could you please clarify this?

An example would be: https://github.com/cta-observatory/ctapipe/blob/master/ctapipe/calib/camera/r1.py#L172

kosack, 2017-12-04:
those classes are sub-classes of `traitlets.config.Application` that asks for a name (which may be different from the class name).  I don't think it's required, and is somewhat confusing.  The only place I've really used it is when writing a command-line tool, to name it the same as the command-line rather than the class (e.g. there may be a Tool class called `MyTool`, but I would set the name to `ctapipe-mytool` so the `--help` message comes out right).  Otherwise, it doesn't seem to be used anywhere. 

See the docs for `traitlets.config` - there are  a number of "standard" class attributes that change the behavior of a traitlets-based class where configuration and provenance is tracked.

kosack, 2017-12-04:
see for example:

http://traitlets.readthedocs.io/en/stable/config.html

dneise, 2017-12-04:
I'm sorry, I do not understand the answer. You say these are all subclasses of `traitlets.config.Application`, but the example I've shown is not a subclass of that:

The example I've shown is this

https://github.com/cta-observatory/ctapipe/blob/89e82e3be4dff7d728fb55213655d199f1e71487/ctapipe/calib/camera/r1.py#L172

and this is a subclass of `traitlets.config.Configurable` not `Application`.

dneise, 2017-12-04:
You further say:

>  there are a number of "standard" class attributes that change the behavior of a traitlets-based class where configuration and provenance is tracked.

I do not find any mention of such standard class attributes in the traitlets documentation. On the page you nicely provided, the only place where a class attribute `name` is mentioned is here

http://traitlets.readthedocs.io/en/stable/config.html#python-configuration-files

The example looks like this:
```python
class MyClass(Configurable):
    name = Unicode(u'defaultname'
        help="the name of the object"
    ).tag(config=True)
    ranking = Integer(0, help="the class's ranking").tag(config=True)
    value = Float(99.0)
    # The rest of the class implementation would go here..
```

but I did not find anything in the text, that says, that `name` plays some kind of special role ... 
..confused....

dneise, 2017-12-04:
Ah I found an example in the `traitlets`github repo:

https://github.com/ipython/traitlets/blob/master/examples/myapp.py

This `Application` also has a class variable called `name`.

I executed it and looked at the `help` to see if the class variable name has any impact on how the help is rendered:

    $ python myapp.py --help-all > with_name.txt
    # Now I commented out line 57 of myapp.py
    $ python myapp.py --help-all > without_name.txt
    $ diff with_name.txt without_name.txt

The diff was empty, so `name` has no impact on how the help looks like.

dneise, 2017-12-05:
Ah by the way, I think this issue is not yet closed. I'd be happy if you could reopen it.

dneise, 2017-12-05:
I have started a little study of traitlets in the context of ctapipe. maybe you'll find this interesting.

https://github.com/dneise/try_traitlets_for_ctapipe

At the moment, I am at "step 10" of this little study, but I might further work on it, if I have time.

kosack, 2017-12-06:
I think name only plays a role in `Application` classes, not `Configurable`.  If we use it in a Configurable or `Component` class, it's probably a mistake and unnecessary.  The only think I can see it doing for the Application class is when using "--help" it prints the name in the resulting output.

kosack, 2017-12-06:
I have to say, the results of your study so far does show that the docs aren't great for the config module of traitlets (some unexpected behavior).  We could always replace the dependence on `traitlest.config.Application` with our own implementation with better checks in the future (still using traitlets classes as the config parameters), or at least submit some bug reports to the developers.  Nevertheless, traitlets is quite nice for flexibility, so I like using them to tell which variables are user-configurable with constraints. That doesn't mean we cannot later change the back-end of how they are written/read, though it would be nice to avoid more code to maintain. 

We had started with other packages like argparse, but they don't support both config-files and command-lines, made it difficult to delegate configuration to sub-classes,  and made it  a bit harder to simply get a dict of all parameters (needed for storage of provenance info).

dneise, 2017-12-06:
> In any case, I suggest we remove the name attribute from any Component classes that currently use one, to avoid confusion, since it seems to be superfluous.

That's nice I'd like to do that.

dneise, 2017-12-06:
> or at least submit some bug reports to the developers. 

yes .. or even try to open some documentation improving PRs, so next time somebody from cta reviews the `traitlets` docu, they have hopefully an easier live. 

However, do you think a dedicated section about traitlets within the ctapipe docu might be okay?

MaxNoe, 2017-12-06:
@dneise I think the whole idea of the configuration, `Component`s and `Tools` should be more explicit. 

Basically any script that handles data needs to be a `Tool` and handle `parent` correctly to save provenance.

closed 2018-02-02 by kosack