cta-observatory/ctapipe #602: Implement VERITAS-style 2-pass trace integration

kosack, 2017-12-11:
We should have an implementation of the 2-pass trace-integration (using timing info) that is described e.g. in *Holder et al, 2016*.  The result is a higher signal-to-noise image, and the ability to apply lower cleaning thresholds in subsequent steps. 

The basic procedure is:

1. use a wide (20ns) trace integration
2. apply a 1st stage image cleaning (tailcuts)
3. calculate the time gradient of the resulting pixels
4. use the time-gradient to predict the position of the pulses to better accuracy 
5. re-integrate with a 10ns window using the information from step 4

This is not so far different from the `NeighborPeakIntegrator` with some extra time information, and would be useful for comparing our results with MARS and EventDisplay.

watsonjj, 2017-12-11:
This is very similar to a charge extraction technique that Heide Costantini implemented in read_hess. I have been intending to re-implement it into ctapipe for a long time.

kosack, 2017-12-11:
@watsonjj yes, I think it's identical in fact (read_hess, MARS, and VERITAS all implemented it). If you have time, that would be great!

GernotMaier, 2017-12-11:
Some points which we learnt using the double pass method:

- use a robust fitter for the determination of the time gradient (a single outlier might otherwise pull the direction of the image axis into the completely wrong direction)
- pass 1 needs reasonable high cleaning levels to determine a good image long axis. But this means that you miss some of the smallest images. Don't discard them - simply use as integration time the typical (average) pulse arrival time
- the assumption is obviously that some calibration has been applied to the pixel arrival times (if necessary for both high and low gain chain)

kosack, 2018-01-09:
Is there a reason why you would do a fit to determine the time gradient rather than just calculating it with a perhaps weighted average? It's essentially the same thing as calculating the moments. You can get rid of outliers the same way as for the moment calculation.

dneise, 2018-01-16:
> 2-pass trace-integration (using timing info) that is described e.g. in Holder et al, 2006. 

For completeness, I think you mean this? 
https://arxiv.org/pdf/astro-ph/0604119.pdf
page 9 ff

dneise, 2018-01-16:
Somebody working on this already?

kosack, 2020-04-13:
just noticed this isn't linked to #1215

HealthyPear, 2020-05-20:
@kosack @watsonjj 

We can now review this discussion in light of what has been merged from #1215 , namely did that PR take into account all what has been discussed here? If yes, I think we can close this issue.

I think that everything is there, aside from improving or adding more fitting algorithms for the timing information (this can be done later I guess).

MaxNoe, 2020-05-20:
I glimpsed over the two-pass integration when doing #1336, there is a lot to be improved still. I'll make an issue with things that catched my eye.

HealthyPear, 2020-05-20:
Ok!

Then I would suggest to make a new one, more general parhaps: I need the one from #1215 to not be changed in order to not bias the comparison with CTA-MARS

MaxNoe, 2020-05-20:
Wait, what?

I'm talking about general improvements to that class! And for comparisons you will hopefully use a fixed version of ctapipe.

We are not doing code freezes, that's what versioning is for.

PR #1215 (2020-05-11): 2-Pass waveform integration à-la-CTA/MARS
Fixes #602 

**Prerequisites:**

- [x] Merge PR #1131 - continuous integration will fail until this is done.

**Description:**

- _ctapipe_-based version of the calibration method used in _CTA/MARS_ and described also [here](https://forge.in2p3.fr/projects/step-by-step-reference-mars-analysis/wiki#Single-pixel-spectrum).
- triggered from development of _protopipe_ and comparison against _CTA/MARS_ (in particular DL1, see 
cta-observatory/protopipe#31 and cta-observatory/protopipe#37 for more information)

**Caveats:**

1. There is at least one cut insi [... cut]
closed 2020-10-26 by maxnoe