cta-observatory/ctapipe #603: problem with running  muon_reconstruction.py on NectarCAM muons

FrancaCassol, 2017-12-11:
Hi, 
wenn I run muon_reconstruction.py on NectarCAM muons it crashes at the "chord_length" function when called by intersect_circle at line 126:

        if self.hole_radius > 0:
            hole_length = self.chord_length(
                self.hole_radius, r / self.hole_radius.value, angle
            )

Indeed in this case (r / self.hole_radius.value)=20=rho in chord_length and chord become negative, which is incompatible with the sqrt performed afterward:

       chord = 1 - (rho * rho * np.sin(phi) * np.sin(phi))
       if rho <= 1:
            chord = radius * (np.sqrt(chord) + rho * np.cos(phi))
        elif rho > 1:
            chord = 2. * radius * np.sqrt(chord)

Can somebody help on that?  I can pass a small simtel_array file if needed.

Thanks
Cheers

Franca

Sojourner92, 2018-01-11:
Hi Franca,
Could you please attach the simtel_array file you mentioned? I'm reasonably sure that the problem is on L173 of muon_reco_functions.py, a fixed (and wrong) mirror hole radius is being passed to the muon line integrator function which results in rho being 20 (when as its meant to be a fractional length it probably shouldn't be greater than 1).

FrancaCassol, 2018-01-11:
Hi Samuel,
yes! I was also guessing something like that, do you know which is the correct mirror hole radius? You can find the simtel_array file in https://www.cppm.in2p3.fr/~cassolb/data/
Thanks!
F.

Sojourner92, 2018-01-12:
Thanks for that. But yes, I'm having trouble finding the correct hole radius for the MSTs. In an ideal world I'd fix this problem for the three SST designs and the LSTs whilst I'm at it; does anyone happen to know what the correct hole radii for those are either?

FrancaCassol, 2018-02-14:
I can say 'yes' for what concerns NectarCAM muons: I could reconstruct them after the fix.
I plan to test muons simulated in other cameras in the coming weeks, but probably this issue can be considered close at the moment.

kosack, 2018-02-14:
Ok, thanks I'll close it. 

Soon, we can add some data files for testing muons... Alison had provided some, but I haven't yet added them to ctapipe-extra since it is growing big, and I think I'll just move test data to a file server in the future.

closed 2018-02-14 by kosack