cta-observatory/ctapipe #611: Why does `EventFileReader` have these `on_changed` methods?

dneise, 2017-12-19:
So far, I often encountered "Reader" classes, which one might call "one-off" or "one-shot" objects. They are not designed to be reused. They have no interface allowing them to be re-used.

E.g. this does not exist:
```python
textfile_reader = open('/some/path/to/file.txt', 'r')
for event in textfile_reader:
    pass

textfile_reader.input_path('/some/other/file.txt')  # <--- does not exist
for event in textfile_reader:
    pass
```

Instead one would simply make a new reader in the second part. A Reader with such an interface is in my understanding lean and simple, it can be explained in half a sentence.

In the context of #610 I was looking at `ctapipe.io.eventfilereader.EventFileReader`, since the guidelines tell us, we need to inherit from `EventFileReader`. So I studied the interface and was wondering why the interface was so big, and in my understanding complicated.

dneise, 2017-12-19:
Another question I had in this context was: Why does every EventFileReader need to have `max_events` and thus copy the same functionality?

Instead of this
```
reader = SpecialReader(path, max_events=10)
for event in reader:
    pass
```

One could think of this

```python
reader = SpecialReader(path)
range_limit = RangeLimit(10)
for event in range_limit(reader):
    pass
```

or in one line:

    for event in RangeLimit(10)(SpecialReader(path)):
        pass

This way, no reader would need to even have a concept of the number of events, or that the number of events could be limited. Especially not each and every child would have to repeat that functionality:

 * [zfits.py](https://github.com/cta-observatory/ctapipe/blob/master/ctapipe/io/zfits.py#L116)
 * [hessio](https://github.com/cta-observatory/ctapipe/blob/master/ctapipe/io/hessio.py#L222)
 * [toymodel](https://github.com/cta-observatory/ctapipe/blob/master/ctapipe/io/toymodel.py#L39)

watsonjj, 2017-12-19:
With regards to the `on_changed` methods:
I created the `EventFileReader` very early on, when the `Component` had just been implemented in ctapipe. The `on_changed` method was something advertised as part of it, so I thought I had best include it. I have never used it though, I create a new reader as you say. Though if someone was to try `EventFileReader.input_path = newpath` then without the `on_changed` it would not load that file. But I don't know if anyone would expect it to. Open to further comments on this...

watsonjj, 2017-12-19:
I think its important max_events it tied with `EventFileReader`. Otherwise you would also have to handle the logic of `EventFileReader.num_events` yourself each time to make sure you stuck to the max_events you want. I see no reason to change it.

dneise, 2017-12-19:
Oh I have another question. You know the command line tool `rm`, right?
```
$ cp --help
Usage: cp [OPTION]... [-T] SOURCE DEST
  or:  cp [OPTION]... SOURCE... DIRECTORY
  or:  cp [OPTION]... -t DIRECTORY SOURCE...
Copy SOURCE to DEST, or multiple SOURCE(s) to DIRECTORY.
```

for me there is a difference between the `OPTION` part and the `SOURCE DEST` part. 
Options are ... well optional .. `cp` works even, when I do not specify them. The `SOURCE DEST` here is not optional, when I do not specify it:
```
$ cp
cp: missing file operand
Try 'cp --help' for more information.
```

I always assumed cta analysis programs to follow a similar pattern. They get a source and a destination. might be on disk, might be a zeromq stream... but these are not optional.

The programs source code defines what should to be done. 

The config file specifies *how* the parts of the analysis should be done, while certain useful defaults are already given. 

-----

Now your answer made me think: Are `input_paths` Options in the opinion of ctapipe?

dneise, 2017-12-20:
I have some more questions:

1.) There might be event streams not coming from a file but from some network, where finding out the number of events is even impossible. How do you plan to handle that?

2.) What is the intended main use case in ctapipe, to iterate over entire files, or to iterate over a certain number of events? I had assumed so far, iterating over a limited number of events is only needed for unit tests. 

3.) I do not understand the connection between `num_events` and `max_events`. 

You say:

> I think its important max_events is tied with EventFileReader. Otherwise you would also have to handle the logic of EventFileReader.num_events yourself each time to make sure you stuck to the max_events you want.

And I am not sure I fully grasp what "the logic of num_events" is which I need to "handle".

I proposed a solution above, which limits the number of iterations, without knowing the maximum possible number of iterations. Let me give a more complete example including tests:
```python
class FakeReader:
    def __init__(self, path):
        ''' path does of course not contain a path, but is simply a
            string defining the  `total_number_of_events`

            This mimics a Reader interface in a simple manner.
        '''
        self.total_number_of_events = int(path)

    def __iter__(self):
        return self

    def __next__(self):
        for i in range(self.total_number_of_events):
            yield i

# First implementation: just a function "around" the reader.
# ----------------------------------------------------------

def limit_range_of_iterable(iterable, limit):
    for i, content in enumerate(iterable):
        if i >= limit:
            break
        yield i

def test_limit_range_of_iterable_on_long_file():
    LIMIT = 10
    reader = FakeReader('1000')

    events = [e for e in limit_range_of_iterable(reader, limit=LIMIT)]
    assert len(events) == LIMIT

def test_limit_range_of_iterable_on_short_file():
    LIMIT = 10
    reader = FakeReader('2')

    events = [e for e in limit_range_of_iterable(reader, limit=LIMIT)]
    assert len(events) <= LIMIT

# Second implementation: Something which feels more like a class
# --------------------------------------------------------------

from functools import partial
def RangeLimit(limit):
    return partial(
        limit_range_of_iterable,
        limit=limit
    )

def test_RangeLimit_on_long_file():
    LIMIT = 10
    reader = FakeReader('1000')
    range_limit = RangeLimit(LIMIT)

    events = [e for e in range_limit(reader)]
    assert len(events) == LIMIT

def test_RangeLimit_on_short_file():
    LIMIT = 10
    reader = FakeReader('2')
    range_limit = RangeLimit(LIMIT)

    events = [e for e in range_limit(reader)]
    assert len(events) <= LIMIT
```

Of course the interface of this "range limiting function" needs to be defined to our liking. Also more tests are in principle needed, to make sure that:

 - the order of the events is not modified by the "range limiter"
 - the events themselves are not modified
 - ...

However, the same is true for all existing implementations of EventFileReaders. They need to show in unit tests, that their `max_events` feature works as expected, individually for every reader, since they all implement this functionality again. Isn't it much easier to implement this functionality only in one place, and also only test it one place, than in many?

dneise, 2017-12-20:
I have another question: What is `num_events` used for in the analysis chain?

watsonjj, 2017-12-20:
Moving this functionality outside of the `EventFileReader` is a hassle for the user. With the current situation, no matter the script, if it uses an `EventFileReader` then the user automatically has the option to specify `max_events` from the command line. Some users will want to use it, some won't. But the writer of the did not have to include any additional object in their script to include this functionality.

Its also an object that only has one purpose, which adds functionality to the reading from an `EventFileReader`, why separate functionality away from the class that uses it?

watsonjj, 2017-12-20:
> I have another question: What is `num_events` used for in the analysis chain?

Some of your suggestions in this thread seem to suggest you are thinking of ctapipe as a processing chain, one input one output and one main purpose. But in reality there are many uses for ctapipe, especially in the prototyping stage.

In the future your line of thinking may be completely valid, but I don't think we are at the stage where we can dismiss the reality that ctapipe is primarily currently used for:
a) Processing files, not streams, and 
b) Performing particular analysis on MC or prototype camera data, not necessarily resulting in the event lists and IRFs (which will be ctapipe's primary use in the future).

I use num_events in almost every ctapipe script I have wrote to analyse our camera's data, its provides the following functionality:
1. I can monitor the progress of my processing of a file, and know how much longer it will take.
2. When performing event-to-event comparisons, I often need to create a numpy array with one dimension the size of the number of events in the file. For example, if I want an array of charges that I can input into a histogram to show the single photoelectron spectrum from low illumination events.

dneise, 2018-01-16:
> I can monitor the progress of my processing of a file, and know how much longer it will take.

During development, I like to do similar things. I like to know how many evts per second I can achieve on a single core. For this I do not need to know the total number of events.

----

>  When performing event-to-event comparisons, I often need to create a numpy array with one dimension the size of the number of events in the file (before I start filling it event-by-event). For example, if I want an array of charges that I can input into a histogram to show the single photoelectron spectrum from low illumination events.

if one does not know the number of events before, one can simply fill a list and then convert that list to an array afterwards. Yes, that is a bit slower, and it needs a bit more mem ... but during development this should not be a problem.

https://stackoverflow.com/a/2106651/1532474

dneise, 2018-01-16:
I wrote on top:

> In the context of #610 I was looking at ctapipe.io.eventfilereader.EventFileReader, since the guidelines tell us, we need to inherit from EventFileReader. So I studied the interface and was wondering why the interface was so big, and in my understanding complicated.

I think the interface got much smaller now, so that #613 solves this issue already now.

closed 2018-01-16 by dneise