cta-observatory/ctapipe #616: Error running muon reconstruction

rlopezcoto, 2018-01-17:
Hi,

when trying to run the muon reconstruction using the example script in:
examples/muon_reconstruction.py

I get the following error:

```sh
/home/rlopez/data_rlopez/cta/ctasoft/ctapipe/ctapipe/image/muon/muon_integrator.py:97: RuntimeWarning: invalid value encountered in sqrt
  chord = 2. * radius * np.sqrt(chord)
/home/rlopez/data_rlopez/cta/ctasoft/ctapipe/ctapipe/image/muon/muon_integrator.py:418: RuntimeWarning: Exception Occured 
fcn is called with following arguments:
           impact_parameter = +4.000000
                        phi = +0.000000
                   centre_x = +0.650597
                   centre_y = -0.706481
                     radius = +1.115991
                 ring_width = +0.100000
    optical_efficiency_muon = +0.100000

  minuit.migrad()

gzip: stdout: Broken pipe
Traceback (most recent call last):
  File "muon_reconstruction.py", line 111, in <module>
    tool.run()
  File "/home/rlopez/data_rlopez/cta/ctasoft/ctapipe/ctapipe/core/tool.py", line 167, in run
    self.start()
  File "muon_reconstruction.py", line 71, in start
    muon_evt = analyze_muon_event(event)
  File "/home/rlopez/data_rlopez/cta/ctasoft/ctapipe/ctapipe/image/muon/muon_reco_functions.py", line 184, in analyze_muon_event
    image[dist_mask])
  File "/home/rlopez/data_rlopez/cta/ctasoft/ctapipe/ctapipe/image/muon/muon_integrator.py", line 416, in fit_muon
    minuit.migrad()
  File "iminuit/_libiminuit.pyx", line 423, in iminuit._libiminuit.Minuit.migrad (iminuit/_libiminuit.cpp:5293)
  File "/home/rlopez/data_rlopez/cta/ctasoft/ctapipe/ctapipe/image/muon/muon_integrator.py", line 299, in likelihood
    self.pixel_y.value,
  File "/home/rlopez/data_rlopez/cta/ctasoft/ctapipe/ctapipe/image/muon/muon_integrator.py", line 249, in image_prediction
    pred *= 0.5 * self.photemit
  File "/home/rlopez/data_rlopez/software/anaconda3/envs/cta-dev/lib/python3.6/site-packages/astropy/units/quantity.py", line 650, in __array_ufunc__
    return self._result_as_quantity(result, unit, out)
  File "/home/rlopez/data_rlopez/software/anaconda3/envs/cta-dev/lib/python3.6/site-packages/astropy/units/quantity.py", line 689, in _result_as_quantity
    out._set_unit(unit)
AttributeError: 'numpy.ndarray' object has no attribute '_set_unit' 
```

The code stops running in line
```sh
/home/rlopez/data_rlopez/cta/ctasoft/ctapipe/ctapipe/image/muon/muon_integrator.py", line 418, in fit_muon
```
I tried changing the values of **chord** not to have negative values, but I still get the same Exception error when running minuit.migrad

MaxNoe, 2018-01-17:
The error implies that something is a plain numpy array where it should be an `astropy.units.Quantiy`:

>
  ```
  'numpy.ndarray' object has no attribute '_set_unit'   
  ```

rlopezcoto, 2018-01-17:
I just realized there is a pull request solving the issue:
#615 
can someone review it so it will be merged?
Thanks!

closed 2018-01-29 by rlopezcoto