cta-observatory/ctapipe #631: Problem with repo after removing large H5 files

kosack, 2018-01-30:
There seems to be a major repo problem right now after I garbage-collected some accidentally committed binary files.  GitHub seems to now think that all forks/branches are completely different to the master and it shows all past commits in the changelog for each PR (not ideal!). 

According to GitHub's help, all forks and PRs just need to be rebased on the current master (git rebase upstream/master), which should be simple, but for me I get a conflict for *all* commits in the history, which would take me weeks to resolve. 

Any git expert have an idea?  I'm now regretting trying to clean up the past commit mistake since this seems to have made it worse.

kosack, 2018-01-30:
I think I'll just revert the master branch changes... So much for cleaning up the repo.

MaxNoe, 2018-01-30:
Yes, revert these changes to the master!

Never, ever touch the master branch of a project that people branched away from. It will be completely impossible to rebase against this new master as **all** conflict resolutions in merges would have to be done again, by everybody.

The only way, I see we could clean up the current master is to merge all currently pending pull requests, then clean up the master, then let all people branch away from the new master.

IMHO there is now way how we can clean up the master now and people will still be able to open pull requests for branches they created before the cleanup.

Force pushing the master with a rebased history is nothing one should do without notice long in advance!

kosack, 2018-01-30:
reverted.  Strangely, I  followed exactly the instructions on the GitHub help page which I would have thought would work.  In any case, we'll just deal with the large repo size.

closed 2018-01-30 by kosack

MaxNoe, 2018-01-30:
Do you mean this page?
https://help.github.com/articles/removing-sensitive-data-from-a-repository/

MaxNoe, 2018-01-30:
> Tell your collaborators to rebase, not merge, any branches they created off of your old (tainted) repository history. One merge commit could reintroduce some or all of the tainted history that you just went to the trouble of purging.

The problem i think with this is, that this will only work, if there were no merge conflicts in between so the rebase is fast forward. Else, one would have to do the conflict resolution time and again.