cta-observatory/ctapipe #642: Container construction is suboptimal due to deepcopy

dneise, 2018-02-13:
First an observation: Creating an empty DataContainer takes ~200us 
```
In [1]: from ctapipe.io.containers import DataContainer
In [2]: %timeit DataContainer()
189 µs ± 2 µs per loop (mean ± std. dev. of 7 runs, 10000 loops each)
```

when I remove the `deepcopy` from `Container.__init__` it speeds up factor of ~100
```
In [2]: %timeit DataContainer()
2.02 µs ± 25.2 ns per loop (mean ± std. dev. of 7 runs, 100000 loops each)
```

Nothing surprising here, `deepcopy` is known to be a costly operation.
Also, I am no fan of premature optimization, which leads to worse code readability or even worse usability.
But when slow code in module A leads to worse usability in module B, I think then some performance optimization, might be in order... or at least, it might be worth to talk about it.

I must admit, that I do not fully grasp, why the `deepcopy` is needed in that place. I would assume, that it is possible to define defaults, which are mutables. And in order to avoid the [mutable default arg gotcha](http://docs.python-guide.org/en/latest/writing/gotchas/#mutable-default-arguments) deepcopy was used here.

I am now taking a closer look at the Container to understand this. Any help is appreciated. Also, in case the same discussion has already been lead, when Container was implemented, and I am wasting time here, please just point me to the result.

Thanks a lot :-D

dneise, 2018-02-13:
Ah I forgot to mention the problem I saw, which in my understanding came from the fact that creating an empty DataContainer is costly.

So I saw this (short version):
```python
def event_source(path):
    container = DataContainer()
    for event in low_level_file_reader(path):
        fill_container_from_event(container, event)
        yield event
```

And then a user of the event source did this:
```python
events = [e for e in event_source(path)]
```

Of course all events where the same.

-----

keeping multiple events, is of course explicitely something we are not planning to do in ctapipe. We are thinking in streams, so there is absolutely no reason for any user to keep a couple of events in a list.

But ... I did anyway. I wanted to play around with them and not always loop over the file again. 

I think it makes code harder to understand, when the result of an operation, in this case an iteration, keeps under ownership of the operand. I currently do not remember a name for this principle, but I think I cam across it in the context of functional programming...

Imagine this:
```python
random = np.random.normal(size = 100)
large = np.where(random > 0)
# look at `large` here

# modify `random` a bit
random += 0.1

# would you expect `large` to have changed at this point?
```

dneise, 2018-02-13:
Ah of course there is another solution for people like me:
```
events = [deepcopy(e) for e in event_source(path)]
```

I simply make a copy of each event I get form the source, to make sure, I have complete ownership. I think however for any passerby reading my code it will be unclear why I made that copy.

That Is the whole reason why I opened this issue....

dneise, 2018-02-13:
So maybe, in hindsight ... I was a bit quick in opening this issue.

MaxNoe, 2018-02-13:
I remember opening or at least commenting on an issue like this long time ago.

Here it is: https://github.com/cta-observatory/ctapipe/issues/69 Actually the issue was about the opposite behaviour and it was closed as "won't fix", kinda funny

dneise, 2018-02-13:
Ah, thanks for finding that.

dneise, 2018-02-13:
I think, I can understand the argument, that this interface is trying to avoid beginners not to fill their memory completely, although I do not agree with it. Let me try to quickly analyze the argument.

### 0. 
This interface succeeds at preventing me from filling my memory entirely when doing this:
```python
events = [e for e in event_source(path_to_huge_file)]
```
(But in addition to not filling my memory, I also do not get want I assumes I'd get. Maybe I wanted to see how many events, fit into mem...)
### 1.
Now let's consider a case where a developers strongly believes, she really needs a list of the last 10 events. She will maybe do something along these lines:
```python
for event in event_source(path_to_huge_file):
    last_ten_events = last_ten_events[-9:].append(event)
    # do something with the last ten events
```
she will test her code, realize it does not work, and override the interface like this:
```python
for event in event_source(path_to_huge_file):
    last_ten_events = last_ten_events[-9:].append(deepcopy(event))
    # do something with the last ten events
```
She might be a bit grumpy over the strange behavior of event_source, but she made her feature pass, and is done with it.

### 2.
Let's assume there are now two processes, which need access to the last 10 events. What happens then?
Both will have to do: `deepcopy(event)` independently from each other, effectively doubling memory footprint. So now what was meant to limit memory waste, has actually caused it.

### 3.
Let's assume analysis is not I/O-bound but CPU bound. Someone might try to spawn several worker threads or processes, which all have a reference to the event_stream, executing `next(event_stream)` to fetch the next event to analyze. This will of course blow up, since the `next()`-call will not only fetch a new event for the current thread, but it will also modify the events in the other processes, while they might still be working on them.

### 4. (subjective and thus weak)
I have the impression this interface is not only leading me to the water but trying to force me to drink (and failing to), while at the same time behaving unintuitively (is that a word?).

dneise, 2018-02-13:
Also, the fact that `Containers` are designed to be modified in place, and reused for new events, mandates the creation of such methods like `Container.reset()`. Standard well-known Python containers, do not have a `reset()` and it is clear why. 

While deleting an single element from a list or dict is immediately plausible (and thus supoported by their interface) resetting an existing list back or dict to its initial state is much easier done by simply creating a fresh one. A python dict has no reset. 

In addition, `reset()` as it is implemented at the moment, seems to be even a tiny little bit slower than just creating a new instance:
```python
In [1]: from ctapipe.io.containers import DataContainer
In [2]: %timeit DataContainer()
186 µs ± 483 ns per loop (mean ± std. dev. of 7 runs, 10000 loops each)
In [3]: container = DataContainer()
In [4]: %timeit container.reset()
191 µs ± 981 ns per loop (mean ± std. dev. of 7 runs, 10000 loops each)
```
Well slower is a bit of a stretch ... let's say it takes the same time ... this is not really surprising.

But `reset()` needs lines of code, not only in Container, but also in `Map`. Also this feature needs to be documented. 

I wonder if it is worth it.

mackaiver, 2018-02-14:
I would find the current behavior very surprising. Which is non optimal I guess. 

> Now let's consider a case where a developers strongly believes, she really needs a list of the last 10 events.

Seems like a reasonable use case to me. I often use a  functional, haskell-like, approach.
    
    take(10, event_source(path_to_file))

The `take` function is described here https://docs.python.org/3/library/itertools.html#itertools-recipes

dneise, 2018-02-14:
My guess at the top was wrong:

>I must admit, that I do not fully grasp, why the deepcopy is needed in that place. I would assume, that it is possible to define defaults, which are mutables. And in order to avoid the mutable default arg gotcha deepcopy was used here.

Max explained to me, that this deepcopy is needed to create all the empty sub-containers of this container.  
So in the current design, there is no way to omit this `deepcopy` there.

dneise, 2018-02-14:
Alternative `core.Container` designs might be possible, but are beyond the scope of this issue. 

I just thought the `deepcopy` could be avoided and thus the Container construction could be sped-up.

I also learned here, that the Container re-use, which I regard as questionable design, was not done because the Container construction is slow, but to help protect users from themselves. 

I think this issue can be closed.

closed 2018-02-14 by dneise