cta-observatory/ctapipe #644: Bug applying cuts for muons?

rlopezcoto, 2018-02-15:
Hi,

I found what seems to be a bug, but wanted to be sure before correcting it.  In l.152 of `muon_reco_functions.py`, there are the two following conditions:

        if(np.sum(pix_im > tailcuts[0]) > 0.1 * minpix
           and np.sum(pix_im) > minpix 

both of them are relating a sum over the content of `pix_im`, that has the pixel containments (pe, I guess). Then it is comparing it to something proportional to `minpix` that is a minimum number of pixels defined in a dictionary. My guess is that the comparison that wanted to be done is something like:

        if(len(pix_im[pix_im > tailcuts[0]]) > 0.1 * minpix
           and len(pix_im) > minpix 

@AMWMitchell, is it alright if I correct it?

dneise, 2018-02-15:
Hey Ruben (no idea of the muon code) but I wanted to say one can quote code directly from the source file like this:
https://github.com/cta-observatory/ctapipe/blob/39cf9af6df8ffed682982aa40a0581d722db76a3/ctapipe/image/muon/muon_reco_functions.py#L152-L161

This way interested noobs like me can quicker find the entire file by following the link and study the context in detail... if they want.

Cheers.

rlopezcoto, 2018-02-15:
thanks for the tip, @dneise!

AMWMitchell, 2018-02-15:
Yes, well spotted - this should be comparing the number of pixels rather than the sum of the charge. Please do correct this, and check if it affects the number passing cuts (shouldn't make a large difference, but there might be fewer events). 
Thanks

rlopezcoto, 2018-02-27:
Solved in #645

closed 2018-02-27 by rlopezcoto