cta-observatory/ctapipe #646: name: get_dataset --> dataset_path?

dneise, 2018-02-15:
I was just made aware of
https://github.com/cta-observatory/ctapipe/blob/39cf9af6df8ffed682982aa40a0581d722db76a3/ctapipe/utils/datasets.py#L109-L127

I think the name of this function is not very well chosen.

There is a difference between a dataset and its path, just like there is a difference between me and my name: a function called `get_dominik` should beam me to your office, the function `get_dominiks_name` simply returns a string. 

Also I think, there is no need to stress the `get` here.

I'd propose: `dataset_path`, if I knew what is meant by dataset here. I assume the name dataset was derived from e.g. `sklearn`s `datasets`? Like it is used in this example?
```python
>>> from sklearn import datasets
>>> iris = datasets.load_iris()
>>> type(iris)
<class 'sklearn.utils.Bunch'>
```

But here one actually gets a specific `Bunch` object full of data for a data analysis task. While `ctapipe.utils.get_dataset` will provide me (not with an object full of data for a data-analysis-task, but with) a path to many different files, for entirely different purposes, like for example this config file here:

```python
In [1]: from ctapipe.utils import get_dataset
In [2]: get_dataset('config.json')
Out[2]: '/home/dneise/.cache/Python-Eggs/ctapipe_extra-0.2.12-py3.6.egg-tmp/ctapipe_resources/config.json'
```

So I'd propose: `resource_file_path()` or so ... 

What do you think?

kosack, 2018-02-19:
agreed. the name is not nice.  Also, there is a `get_table` which really does open the dataset and return a table, so this imples the name of `get_dataset` should at least be `get_dataset_path` or perhaps `get_dataset_url` since it may be a remote download.

kosack, 2018-02-19:
Pretty soon we will have a server where we can have test files with *https* access, so it would be nice if we can hide the downloading of test data behind this call as well, like how astropy does it (where it downloads and caches the file if it is not there locally, and otherwise will return the cached version).

PR #704 (2018-04-12): Refactor datasets
- changed get_dataset -> get_dataset_path (old one is still there but deprecated, so shouldn't break anything), fixes #646 

- added a `get_structured_dataset` helper function that works like `get_table_dataset` but opens a YAML or JSON file and returns the dictionary of values (useful for config files where we use one of those formats)
commit 9e03b339 (2018-04-12): Refactor datasets (#704)

* minor refactoring of datasets

- changed get_dataset -> get_dataset_path (fixes #646)
- added get_structured_dataset() that works like get_table_dataset, but for YAML or JSON files.

* added unit test for get_structured_dataset
closed 2018-04-12 by kosack