cta-observatory/ctapipe #647: Muon reconstruction (optical efficiency)

vasylbeshley, 2018-02-15:
I am using the ctapipe for calculations of the muon ring parameters. But optical efficiency calculated for SST-1M:DigiCam is very small - above 0.01 (see image below). I have used muon_integrator.
[image]

I have also used the old version of the ctapipe (November 2016) to compare the results. In this case the optical efficiency is ~ 0.1 (good for SST-1M: DIgiCam)
[image]

So, the optical efficiency for the old version is better than calculated efficiency by the new version of the ctapipe
Does somebody know what the problem is?

kosack, 2018-03-07:
@rlopezcoto do you have an idea? Could this have been affected by a recent change?

rlopezcoto, 2018-03-07:
No, idea. I guess what is plotted there is the container `optical_efficiency_muon `, right? This was there before I started committing any changes (see slide 8 of https://indico.cta-observatory.org/event/1339/contributions/4126/attachments/3513/3968/MuonCtapipeApr17.pdf)
@AMWMitchell, any idea why this optical efficiency is so low?

vasylbeshley, 2018-03-08:
Hi all!
I have compared values of x(deg) and y(deg) coordinates between two versions of the ctapipe (the old (optical efficiency ~ 0.1) and the new (optical efficiency ~ 0.01). )

x(m) and y(m) coordinates are identical for both versions (muon_reco_function.py)

!new
x m =  [ 0.48370001  0.46327001  0.46327001 ... -0.45644999 -0.47689 -0.47689   ] m

y m=  [-0.0118 -0.0472 -0.0236 ...  0.0354 -0.0236  0.    ] m
camera_coord <CameraFrame Coordinate (focal_length=5.599999904632568 m, rotation=30d00m00s, pointing_direction=None, array_direction=None): (x, y) in m
    [( 0.48370001, -0.0118), ( 0.46327001, -0.0472),
     ( 0.46327001, -0.0236), ..., (-0.45644999,  0.0354),
     (-0.47689   , -0.0236), (-0.47689   ,  0.    )]>

!old
x m [ 0.48370001  0.46327001  0.46327001 ..., -0.45644999 -0.47689    -0.47689   ] m

y m [-0.0118 -0.0472 -0.0236 ...,  0.0354 -0.0236  0.    ] m
camera_coord <CameraFrame Coordinate (focal_length=5.599999904632568 m, rotation=0.0 deg, pointing_direction=None, array_direction=None): (x, y, z) in m
    [(0.48370001, -0.0118, 0.0), (0.46327001, -0.0472, 0.0),
     (0.46327001, -0.0236, 0.0), ..., (-0.45644999, 0.0354, 0.0),
     (-0.47689, -0.0236, 0.0), (-0.47689, 0.0, 0.0)]>

but for x and y coordinates in degrees the values are different
 
!new
 x deg=  [ 4.34625834  4.34633101  4.22560061 ... -4.22553601 -4.10482167 -4.22555207] deg
 
y deg=  [ 2.36990596  1.95172555  2.16083674 ... -2.021392   -2.64873481 -2.43962363] deg

!old
x deg [ 4.94892311  4.73989584  4.73989584 ..., -4.67011753 -4.87924725 -4.87924725] deg

y deg [ -1.20730399e-01  -4.82921597e-01  -2.41460798e-01 ...,   3.62191178e-01 -2.41460798e-01   5.97535453e-16] deg

I think that the problem can be in the coordinate transformations. How to check that NominalFrame and HorizonFrame classes work correctly?

dneise, 2018-03-09:
> How to check that NominalFrame and HorizonFrame classes work correctly?

Maybe there are some tests, one could add to [`coordinates/tests/test_coordinates.py`](https://github.com/cta-observatory/ctapipe/blob/master/ctapipe/coordinates/tests/test_coordinates.py) in order to find out if everything works as expected?

ParsonsRD, 2018-04-23:
Hi, the reason for the difference here is not due to changes in the coordinates classes. But if you look at the output you posted here

NEW
camera_coord <CameraFrame Coordinate (focal_length=5.599999904632568 m, **rotation=30d00m00s**, pointing_direction=None, array_direction=None): (x, y) in m

OLD
camera_coord <CameraFrame Coordinate (focal_length=5.599999904632568 m, **rotation=0.0 deg**, pointing_direction=None, array_direction=None): (x, y, z) in m

You see that the camera rotation angle has changed in between these versions from 0 deg to 30 deg. Which one of these is the correct one I can't say, it depends how the pixel positions are defined in the sim_telarray sims. But I would guess it is 0.

kosack, 2018-04-23:
>  But I would guess it is 0.

The rotation angle is currently "de-rotated" by the `camera_rot` angle when we construct the `CameraGeometry`. I'm not sure that's the best place to do it, but it's where it is so far, which I would guess you are correct - the rotation should simply be 0.

AMWMitchell, 2018-06-06:
Hi All,
I'm not sure why the normalisation changed, but I'm pretty sure we're missing a normalisation factor taking into account the number of pixels falling along the muon ring, i.e. of order (angular pixel width)/ (2*pi*muon ring radius).
Should be added to the likelihood function; I'm not sure about the angular pixel width for arbitrary directions across a square pixel though?

maxnoe, 2020-10-26:
This should be fixed now, at least with LST, we are in the correct ballpark. If there are issues for specific telescopes, please open a new issue.

closed 2020-10-26 by maxnoe