cta-observatory/ctapipe #657: Cleanup and structure examples

MaxNoe, 2018-02-22:
I think we have too many examples in the base folder.

My proposol would be to have 3-5 basic examples that show the very basics to get started in the base folder and move anything else to subfolders to make them easier to find for people who look for certain things.

Right now, any newcomer won't really know where to start.

kosack, 2018-03-07:
Sounds good. I think we also need to clean up those remaining examples as well.  The other question is how many should be notebooks vs python code (notebook are nicer for demos, but harder to test).

kosack, 2018-03-07:
Many of the examples are simply plotting examples, so I suggest we make a plotting subdir, and perhaps remove the very old "brainstorm" and "obselete" directories.  The other option is to separate the examples into another package (like ctapipe-extras) but that may be confusing.

kosack, 2018-10-29:
see also #739 : should have core examples as notebooks and have them render in Sphinx (as in GammaPy)

kosack, 2019-01-29:
fixed for notebook examples in #941  Still have to do this for regular examples.

PR #2266 (2023-03-30): Remove or move examples to documentation
fixes #931
fixes #657

- improved the CameraDisplay and ArrayDisplay example notebooks in the docs
- moved some code from the old examples into notebook
- removed all old script examples
closed 2023-03-30 by maxnoe